dotnet-bluetooth-le/dotnet-bluetooth-le #322: Device disconnected when perform characteristic.WriteAsync with a long byte array

kelvin195eclipse, 2018-08-08:
I am working with a Thermal Receipt Printer using this plugin, the characteristic.WriteAsync method everything works great with Android, but iOS only works over line-by-line printing using WriteAsync, but if I write a long byte array (Entire Receipt) on iOS, it shows that the device is disconnected.

## Steps to reproduce

1. Create a long byte array containing up to 10 lines of text.

2. Deploy the app on iOS phones.

2. Perform characteristic.WriteAsync to the byte array to the printer.

3. App crash with the error "Device Disconnected".

## Expected behavior
Print out the entire receipt.

## Actual behavior
App crashes with device disconnecting error.

### Crashlog
Device 876f8c2f-3e31-dfbe-3c6d-599f55f88853 disconnected while writing characteristic with 49535343-8841-43f4-a8d4-ecbe34729bb3.

## Configuration

Version of The Plugin: 2.0.0-pre1

Platform: iOS 11.4.1

Device: All iPhones, Mustek Thermal Printer MK380

flyingxu, 2018-08-13:
What if you slice your long data array into Multiple WriteAsync ?

asj-it, 2018-10-11:
I have found that this issue happens only in v2.0.0-pre1 and not the stable v1.3 version.  I have rolled back to v1.3 because of this.
I was sending very small writes ( literally Wrote "Hi") and it caused the exception