dotnet-bluetooth-le/dotnet-bluetooth-le #325: after the  ConnectToKnownDeviceAsync is not reading the data on Android versions greater than 8

urmatgit, 2018-09-09:
## Steps to reproduce

1. Connect to device
1.1 find characteristic
1.2 run StartUpdatesAsync
2. Write to device 
3. Disconnect then wait 10 second
4. Re-connect with ConnectToKnownDeviceAsync
4.1.  find characteristic
4.2. run StartUpdatesAsync
 
## Expected behavior
Not reading data after reconnect to device 

## Actual behavior
continue reading data

### Crashlog

 

## Configuration

Version of the Plugin:** 1.3.0, 2.0.0-pre1

Platform: Android 8

Device: All

fryette, 2019-05-29:
experienced the same issue. Are u found workaround?

PawelGasiorowski, 2019-06-26:
I have the same problem. Anyone can help?

I connect to the device (pairing enabled)
I immediately get Connection Lost event
I can't get services - I get disconnected while fetching services error but it is only a result of a problem that is causing the connection losse.

fryette, 2019-06-26:
Switched to shiny
https://github.com/shinyorg/shiny

xabre, 2019-08-04:
Did you try the sample app from the master branch to try to reproduce this?