dotnet-bluetooth-le/dotnet-bluetooth-le #326: NativeDevice.Rssi, State and Id properties not found

asierdanjou, 2018-09-25:
In the Xmal code:

                                _<Label
                                    FontAttributes="Bold"
                                    Text="{Binding NativeDevice.Name}">
                                </Label>
                                <Label
                                    Text="{Binding NativeDevice.Rssi}">
                                </Label>
                                <Label
                                    FontAttributes="Italic"
                                    Text="{Binding NativeDevice.Address}">
                                </Label>
                                <Label
                                    FontAttributes="Bold"
                                    Text="{Binding NativeDevice.State}">
                                </Label>_

But in the output window:

_[0:] Binding: 'Rssi' property not found on '00:0B:57:EF:2D:90', target property: 'Xamarin.Forms.Label.Text'
[0:] Binding: 'State' property not found on '00:0B:57:EF:2D:90', target property: 'Xamarin.Forms.Label.Text'_

When I'm tiping code nativeDevice property (IDevice) it shows me State and Rssi options but then when executing it does not find these properties. Any helping ?

MKuckert, 2019-03-08:
Is a linker in your project enabled?

xabre, 2019-08-04:
Ensure your binding sources are correct. Check the sample project. Works fine,

closed 2019-08-04 by xabre