dotnet-bluetooth-le/dotnet-bluetooth-le #330: About GetCharacteristicAsync returning null

hung1xpro, 2018-10-15:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

```
   var service = await deviceList[count].GetServiceAsync(Guid.Parse("xxx"));
   var characteristic = await service.GetCharacteristicAsync(Guid.Parse("xxx"));
                                    
```

## Expected behavior
It should return characteristic value

## Actual behavior
service  return value can use, but characteristic can't, it still return null so i stuck ...
var characteristic = await service.GetCharacteristicAsync(Guid.Parse("xxxx")); still return null;
                                   

### Crashlog
 GetCharacteristicAsync return null;
If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:** Android 8.0

**Device:** Note 8

hung1xpro, 2018-10-15:
By the way, my GATT server is:

`
 Xamarin.Forms.Device.BeginInvokeOnMainThread(async () =>
            {
                server = CrossBleAdapter.Current.CreateGattServer();
                Guid newa = Guid.Parse("xxxxxx");// Guid.NewGuid();
                var service = server.AddService(newa, true);
                newa = Guid.Parse("xxxxx);
                var characteristic = service.AddCharacteristic(
                    newa,
                    CharacteristicProperties.Read | CharacteristicProperties.Write | CharacteristicProperties.WriteNoResponse,
                    GattPermissions.Read | GattPermissions.Write
                );
                            var notifyCharacteristic = service.AddCharacteristic
                            (
                                Guid.Parse("xxxxx"),
                                CharacteristicProperties.Indicate | CharacteristicProperties.Notify,
                                GattPermissions.Read | GattPermissions.Write
                            );

                            IDisposable notifyBroadcast = null;
                            notifyCharacteristic.WhenDeviceSubscriptionChanged().Subscribe(ex =>
                            {
                                var @event = ex.IsSubscribed ? "Subscribed" : "Unsubcribed";
                                des.Text += @event.ToString() + "\n";
                                if (notifyBroadcast == null)
                                {

                                }
                            }); 

                characteristic.WhenReadReceived().Subscribe(x =>
                {
                    var write = "HELLO";

                    // you must set a reply value
                    x.Value = Encoding.UTF8.GetBytes(write);

                    x.Status = GattStatus.Success; // you can optionally set a status, but it defaults to Success
                });
                characteristic.WhenWriteReceived().Subscribe(x =>
                {
                    var write = Encoding.UTF8.GetString(x.Value, 0, x.Value.Length);
                    // do something value
                });

                await server.Start(new AdvertisementData
                {
                    LocalName = "GATT SERVER"
                });

            });

`

xabre, 2019-08-11:
Try it out with the sample app. Report if it works there.

closed 2019-08-11 by xabre