dotnet-bluetooth-le/dotnet-bluetooth-le #334: Device Name Returning Null

Connor2hd, 2018-10-31:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

Sorry I can't post the source code, it's for work.

1. Scan for Bluetooth Devices

2. Successfully (kind of) receive devices.

3. See they have an Address, but no name.

## Expected behavior
I should be able to see the device name

## Actual behavior
The device name on each device object is null

### Crashlog

Here is the code I am using for scanning.  It works fine, permissions are fine, it just doesn't return a device name.

                    nativeDeviceList.Clear();
                    adapter.ScanMode = Plugin.BLE.Abstractions.Contracts.ScanMode.Balanced;
                    adapter.ScanTimeout = 10000;
                    adapter.DeviceDiscovered += (obj, a) =>
                    {
                        if (!nativeDeviceList.Contains(a.Device))
                            nativeDeviceList.Add(a.Device);
                    };
                    await adapter.StartScanningForDevicesAsync();

## Configuration

**Version of the Plugin:** 1.3

**Platform:** Android 6.1

**Device:** LG G5

xabre, 2018-11-16:
@Connor2hd Please a thrird party BLE scanner app to see how the same devices appear there.

mjspk, 2018-12-24:
same problem here not just the device name even the id any ideas

Abdulsametileri, 2019-03-05:
Is there any update? I have same problem.

dr1sshamra, 2019-04-03:
any updates guys ? the same problem here

xabre, 2019-08-05:
This is not really an issue as your BLE device might not be advertising any device name. You can double check this with a 3rd party scanner app like nRF Connect.

danielleiszen, 2020-03-07:
this is simply ridiculous.  for two years this issue still persists. is this that rare that people list the devices? or no one cares? or is it that complicated? I thought it was an obvious and quite everyday task. and if that so, what good is a library that does not provide convenient and intuitive way of handling simple things?

rahuldhole, 2020-04-01:
Did anyone get a solution?

BaseCrusher, 2020-04-30:
Can also confirm that.
I'm having this problem on Android only. On iOS I have never encountered this issue. 
It's not persistent.

It seems like as soon as I started another app that is using the BLE scanner it worked again. Not sure if it has anything to do with it. Might have something to do with the scanner mode.

I'll report again when I have more infromation.

steeve49, 2020-06-02:
Same problem here.
I activated the location permission, I see new devices appearing. Some of them are named, others return null name.
I don't understand why the same devices appear correctly named into my device bluetooth parameter's page and not in the application.
I also tried the app sample and nRF Connect => same problem. Two devices are recognized, the others not.

BaseCrusher, 2020-06-02:
> Same problem here.
> I activated the location permission, I see new devices appearing. Some of them are named, others return null name.
> I don't understand why the same devices appear correctly named [...]

Most of BLE devices donot advertise their name. So if your devices do have a name and those are displayed there is no problem. My observation was, that nRF Connect is working flawlessly. So I would take a look at your BLE devices. ;)

In my case I have one BLE device which I use to connect to the app for iOS and Android. On iOS it always works without any issues. On Android I encounter it sometimes. Though I haven't encountered it now for a long time. 

It might have something to do on which thread the app is processing the advertisement data. But I'm not sure. :/

steeve49, 2020-06-02:
@BaseCrusher : Thanks for your answer. What I don't understand is why my phone is able to display the names and not de BT LE library.
Perhaps it's not an LE device but I tried to find a portable library with classic bluetooth and didn't managed to find it... 
I'm stuck trying at least displaying a list of BT devices (LE or not) into a portable xamarin forms application library and it's reaaaaaally painful...

BaseCrusher, 2020-09-07:
So I encountered this issue again today. 
Same informations as there where before. But atleast ive gotten the Crashlog. though I don't know how helpful it is.

**System.NullReferenceException:** 'Object reference not set to an instance of an object.'
09-07 10:08:21.785 D/Mono    (28821): DllImport attempting to load: '/system/lib64/liblog.so'.
09-07 10:08:21.786 D/Mono    (28821): DllImport loaded library '/system/lib64/liblog.so'.
09-07 10:08:21.786 D/Mono    (28821): DllImport searching in: '/system/lib64/liblog.so' ('/system/lib64/liblog.so').
09-07 10:08:21.786 D/Mono    (28821): Searching for '__android_log_print'.
09-07 10:08:21.786 D/Mono    (28821): Probing '__android_log_print'.
09-07 10:08:21.786 D/Mono    (28821): Found as '__android_log_print'.
09-07 10:08:21.792 I/MonoDroid(28821): UNHANDLED EXCEPTION:
09-07 10:08:21.796 I/MonoDroid(28821): System.NullReferenceException: Object reference not set to an instance of an object.
09-07 10:08:21.796 I/MonoDroid(28821):   at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__7_0 (System.Object state) [0x00000] in /Users/builder/jenkins/workspace/archive-mono/2020-02/android/release/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/AsyncMethodBuilder.cs:1021 
09-07 10:08:21.796 I/MonoDroid(28821):   at Android.App.SyncContext+<>c__DisplayClass2_0.<Post>b__0 () [0x00000] in <9a14097a65a445eaa85f6a3a1ade52a3>:0 
09-07 10:08:21.796 I/MonoDroid(28821):   at Java.Lang.Thread+RunnableImplementor.Run () [0x00008] in <9a14097a65a445eaa85f6a3a1ade52a3>:0 
09-07 10:08:21.796 I/MonoDroid(28821):   at Java.Lang.IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this) [0x00008] in <9a14097a65a445eaa85f6a3a1ade52a3>:0 
09-07 10:08:21.796 I/MonoDroid(28821):   at (wrapper dynamic-method) Android.Runtime.DynamicMethodNameCounter.1(intptr,intptr)
09-07 10:08:21.814 D/Mono    (28821): DllImport searching in: '__Internal' ('(null)').
09-07 10:08:21.814 D/Mono    (28821): Searching for 'java_interop_jnienv_throw'.
09-07 10:08:21.814 D/Mono    (28821): Probing 'java_interop_jnienv_throw'.
09-07 10:08:21.815 D/Mono    (28821): Found as 'java_interop_jnienv_throw'.

**System.NullReferenceException:** 'Object reference not set to an instance of an object.'
09-07 10:08:22.807 E/mono    (28821): 
09-07 10:08:22.807 E/mono    (28821): Unhandled Exception:
09-07 10:08:22.807 E/mono    (28821): System.NullReferenceException: Object reference not set to an instance of an object.
09-07 10:08:22.807 E/mono    (28821):   at (wrapper dynamic-method) Android.Runtime.DynamicMethodNameCounter.1(intptr,intptr)
09-07 10:08:22.807 E/mono    (28821):   at (wrapper native-to-managed) Android.Runtime.DynamicMethodNameCounter.1(intptr,intptr)
09-07 10:08:22.809 E/mono-rt (28821): [ERROR] FATAL UNHANDLED EXCEPTION: System.NullReferenceException: Object reference not set to an instance of an object.
09-07 10:08:22.809 E/mono-rt (28821):   at (wrapper dynamic-method) Android.Runtime.DynamicMethodNameCounter.1(intptr,intptr)
09-07 10:08:22.809 E/mono-rt (28821):   at (wrapper native-to-managed) Android.Runtime.DynamicMethodNameCounter.1(intptr,intptr)