dotnet-bluetooth-le/dotnet-bluetooth-le #338: Release v2.0.0-pre2 to nuget

flyingxu, 2018-12-16:
The current 2.0.0 branch has some update from v2.0.0-pre1, the main update is the removal of MvvmCross5 and compatibility of MvvmCross6.

It's actually a good time to release it to nuget as pre2. Currently, I have to locally compile it and then use it with a MvvmCross  6 project.

flyingxu, 2018-12-18:
@xabre @smstuebe

JPSiller, 2019-01-25:
+1 BLE functionality broken after upgrading to latest MvvmCross.

JPSiller, 2019-01-29:
Hi @flyingxu Do you have an estimate when this will be released to nuget? 

Thanks

ahi, 2019-07-23:
It would be nice to see the changes from [826e066](https://github.com/xabre/xamarin-bluetooth-le/commit/826e066d70f3e1b561340a3193fbdb0b0df77a20) in nuget. Are there any plans for a new release?

xabre, 2019-07-31:
Will do a release this weekend hopefully :)

xabre, 2019-08-04:
published 2.0.1 and 2.1.0-pre.1

closed 2019-08-04 by xabre