dotnet-bluetooth-le/dotnet-bluetooth-le #342: How to get the response from BLE device after characteristic.WriteAsync(data)

wangguan1888, 2019-02-18:
Hello.

Now, I'm trying to write characteristic and get the response from BLE device.
After connecting to BLE device, I have written data to BLE device. but I don't know how to receive the result from BLE device.
 
## Steps to reproduce
1.var service = await device.GetServiceAsync(Guid.Parse(My ID));                         //Get Service
2.var characteristic = await service.GetCharacteristicAsync(Guid.Parse(My ID));    //Get Characteristic
3.var sendcommand = await characteristic.WriteAsync(My Data);                         //Write Data
4.Get response from BLE device (like onCharacteristicWrite , characteristic.getValue() in Java)                     //Receive value

## Expected behavior
I would like to get the response.

## Actual behavior
I don't know how to get response.

### Crashlog

## Configuration
**Version of the Plugin:2.0.0-pre1
**Platform: iOS 12.1 / Android 7.1
**Device: iPhone XR / Asus Android

closed 2019-03-04 by wangguan1888

arenalorenzo, 2019-04-12:
Did you find a solution? I'm having the same issue.

MohamadAminSoltani, 2019-05-17:
I Also have the same problem :(

lorenzo-arena, 2019-05-17:
I managed to get to a solution: (FYI, in my case I'm only interested in communicating using one characteristic) after connecting to a device I retrieve the characteristics list and then attach a delegate to the `ValueUpdated` event

```cs
public async Task<bool> ConnectToDeviceAsync()
{
    var ble = CrossBluetoothLE.Current;
    var adapter = CrossBluetoothLE.Current.Adapter;
    List<IDevice> deviceList = new List<IDevice>();

    adapter.DeviceDiscovered += (s, a) => deviceList.Add(a.Device);
    await adapter.StartScanningForDevicesAsync();

    IDevice device = deviceList.Find(x => x.Name == _deviceName);

    if(device == null)
        return await Task.FromResult(false);

    try
    {
        await adapter.ConnectToDeviceAsync(device);
        
        // Be sure to get the characteristic you're interested in
        List<IService> services = (List<IService>)await device.GetServicesAsync();
        List<ICharacteristic> characteristics = (List<ICharacteristic>)await services[1].GetCharacteristicsAsync();
        var myChar = characteristics.Find(x => x.Uuid == _analyzerCharacteristic);

        myChar.ValueUpdated += OnCharacteristicValueUpdated;

        // Calling this method start the char monitoring
        await myChar.StartUpdatesAsync();

        await GetDeviceInfo();       
    }
    catch (Exception e)
    {
        await adapter.DisconnectDeviceAsync(device);
        throw e;
    }

    return await Task.FromResult(true);
}
```

Where the `OnCharacteristicValueUpdated` is a method like

```cs
protected virtual void OnCharacteristicValueUpdated(object o, CharacteristicUpdatedEventArgs args)
{
    // _responseFrame is a byte[] class property
    _responseFrame = args.Characteristic.Value;
}
```

Then you can write using `await myChar.WriteAsync(frameToSend);`, wait using `Task.Delay(someMs);` and retrieve the response from `_responseFrame`

suriyaKalidoss21, 2019-09-18:
Hi @lorenzo-arena ,
  I have two characteristic one have written permission others have update permission. It's
[image]
 not possible to write and read using same characteristic how do you do it?. Could you please provide some solution

suriyaKalidoss21, 2019-09-18:
did anybody have a solution?