dotnet-bluetooth-le/dotnet-bluetooth-le #345: How to use characteristic.ValueUpdated and characteristic.StartUpdatesAsync()

wangguan1888, 2019-03-04:
Hello.

I'm new to xamarin forms and C#. 
Now, I'm trying to write characteristic and get the response from BLE device.
After sending command to BLE device, I would like to use characteristic.ValueUpdated to get response.
But my code doesn't work. So I would like to know how to use characteristic.ValueUpdated and characteristic.StartUpdatesAsync().

## Steps to reproduce
1. Connect BLE device and get Device, Service and Characteristic
2. Use await characteristic.WriteAsync(command) to send command to BLE device
3. Use characteristic.ValueUpdated and await characteristic.StartUpdatesAsync() to get response from BLE device.

## Expected behavior
The event characteristic.ValueUpdated shall be called and I can get the response.

## Actual behavior
characteristic.ValueUpdated is not be called.

### Crashlog
None

## Configuration
**Version of the Plugin:2.0.0-pre1
**Platform: iOS 12.1 / Android 7.1
**Device: iPhone XR / Asus Android

wangguan1888, 2019-03-04:
My code is here.
I would like to get args.Characteristic.Value, but this code doesn't work.

        public int SendCommand(byte[] command)
        {
            if (device == null || service == null || characteristic == null)
            {
                return 1;
            }

            var result1 = WriteCharacteristic(command);

            receive_data = GetResponse().Result;

            if (receive_data == null || receive_data[1] != 0x00)
            {
                return 1;
            }

            return 0;
        }

        private async Task<bool> WriteCharacteristic(byte[] command)
        {
            await characteristic.WriteAsync(command);
            return true;
        }

        private async Task<byte[]> GetResponse()
        {
            byte[] bytes = new byte[20];

            characteristic.ValueUpdated += (o, args) =>
            {
                bytes = args.Characteristic.Value;
            };

            await characteristic.StartUpdatesAsync();

            return bytes;
        }

MKuckert, 2019-03-08:
You're intermixing async and sync code

closed 2019-03-11 by wangguan1888

suriyaKalidoss21, 2019-09-18:
Hi @wangguan1888  did this issue resolved??
`  public async Task<int> SendCommand(byte[] command,
ICharacteristic writechar,ICharacteristic readchar)
        {
           
 byte[] receive_data = new byte[20];
            var result1 = await WriteCharacteristic(command, writechar);
            receive_data = await GetResponse(readchar);

            if (receive_data == null || receive_data[1] != 0x00)
            {
                return 1;
            }

            return 0;
        }

        private async Task<bool> WriteCharacteristic(byte[] command,ICharacteristic writechar)
        {
            await writechar.WriteAsync(command);
            return true;
        }

        private async Task<byte[]> GetResponse(ICharacteristic readchar)
        {
            byte[] bytes = new byte[20];

            readchar.ValueUpdated += (o, args) =>
            {
                bytes = args.Characteristic.Value;
            };

            await readchar.StartUpdatesAsync();

            return bytes; here i don't get the bytes
        }`