dotnet-bluetooth-le/dotnet-bluetooth-le #351: GattCallback.OnConnectionStateChange Null Pointer Exception

devopsone, 2019-04-04:
The Android side of my app is receiving lots of crashes which i cannot fix in the app. Its not a real bug in the lib but its need a workaround which Google already mentioned. From my understanding this NPE can happen if disconnect() is issued and immediately close() is called. They say close() must be called in the OnConnectionStateChange(). As another user points out this also helps to get rid of the Gatt 133 Error. Does the lib calls close() anywhere? I can only find disconnect() yet.

## Steps to reproduce

https://issuetracker.google.com/issues/37057260

https://stackoverflow.com/questions/29758890/bluetooth-gatt-callback-not-working-with-new-api-for-lollipop

## Expected behavior

Crashes should not occur. The workaround from Google should fix it.

## Actual behavior

Crashes App

### Crashlog

Xamarin Exception Stack:
System.NullReferenceException: Object reference not set to an instance of an object
  at System.Collections.Generic.Dictionary`2[TKey,TValue].TryInsert (TKey key, TValue value, System.Collections.Generic.InsertionBehavior behavior) [0x00111] in <e549bf8ac8284248b468e8721bb03e4f>:0
  at System.Collections.Generic.Dictionary`2[TKey,TValue].set_Item (TKey key, TValue value) [0x00000] in <e549bf8ac8284248b468e8721bb03e4f>:0
  at Plugin.BLE.Android.GattCallback.OnConnectionStateChange (Android.Bluetooth.BluetoothGatt gatt, Android.Bluetooth.GattStatus status, Android.Bluetooth.ProfileState newState) [0x0028e] in <1dca03547a2544928ae0f0edf98e3a73>:0
  at Android.Bluetooth.BluetoothGattCallback.n_OnConnectionStateChange_Landroid_bluetooth_BluetoothGatt_II (System.IntPtr jnienv, System.IntPtr native__this, System.IntPtr native_gatt, System.Int32 native_status, System.Int32 native_newState) [0x00016] in <5d6a6ccead1546c7ad1d93da407321b7>:0
  at (wrapper dynamic-method) System.Object.87(intptr,intptr,intptr,int,int)
## Configuration

**Version of the Plugin:**

1.3.0

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)

Starting with Lollipop

**Device:** e.g. HTC Sensation /i Phone 7 ...

Too Many

kikutano, 2019-07-05:
@devopsone this bug is really annoying, did you found some workaround to fix it? Thanks.

devopsone, 2019-07-08:
> @devopsone this bug is really annoying, did you found some workaround to fix it? Thanks.

Sorry I didn't find a workaround.

xabre, 2019-08-05:
Feel free to reopen if still an issue in v2.0.1

closed 2019-08-05 by xabre