dotnet-bluetooth-le/dotnet-bluetooth-le #352: Crash on Device disconnected while fetching services on Android 9.0

tqngan1984, 2019-04-05:
This issue happened in Fabric

[image]

### Crashlog

Non-fatal Exception: java.lang.Throwable: Device Perifit(o) disconnected while fetching services.
       at Plugin.BLE.Abstractions.Utils.TaskBuilder+<FromEvent>d__1`3[TReturn,TEventHandler,TRejectHandler].MoveNext ()(filename unknown)
       at Plugin.BLE.Android.Device+<GetServicesNativeAsync>d__14.MoveNext ()(filename unknown)
       at Plugin.BLE.Abstractions.DeviceBase+<GetServicesAsync>d__27.MoveNext ()(filename unknown)
       at Plugin.BLE.Abstractions.DeviceBase+<GetServiceAsync>d__28.MoveNext ()(filename unknown)

## Configuration

**Version of the Plugin:** 1.3.0

**Platform:** Android 9.0

**Device:** Motorola Moto G6

PawelGasiorowski, 2019-06-26:
I have the same problem. Anyone can help?
1. I connect to the device (pairing enabled)
2. I immediately get Connection Lost event
3. I can't get services - I get disconnected while fetching services error but it is only a result of a problem that is causing the connection losse.

MarcAlx, 2019-07-26:
Confirm this is an issue, it also happens to me (not only on Android 9.0 but also on 7.0).

My step to repoduce : 

1. connect to device
2. Unplug device
3. try to access device

in my case the connection lost event hasn't been fired (while on iOS it is). 
What's blocking to me it's that the exception thrown can't be catch, leading to an app crash.

xabre, 2019-08-05:
Feel free to reopen if this is still an issue in v2.0.1.

closed 2019-08-05 by xabre