dotnet-bluetooth-le/dotnet-bluetooth-le #354: StartUpdatesAsync() / ValueUpdated not called

Joden31, 2019-04-17:
Hi,
The component I want to connect to works as follows:
We write in a characteristic (TX)
and we come to read the answer in another characteristic (RX).

The RX characteristic has the following attributes:
canread: false
canwrite: false
cannotify: true
So I can not do it by simple reading.
I tried the following code for an event to be called when the characteristic is changed, but the program never goes into this event.

I have tried many solutions already proposed in the previous discussions but none has worked for now.
However I am beginner in xamarin, Maybe I did something wrong.

```
this.Adapter.DeviceConnected += async (s, a) =>
            {
                Xamarin.Forms.Device.BeginInvokeOnMainThread(new Action(async () =>
                {
                    this.Connected = true;
                    this.Not_Connected = false;
                    this.Liste_Device_Connected.Add(a.Device);

                    this.Service_Selected = await this.Device_Selected.GetServiceAsync(Guid.Parse("6E400001-B5A3-F393-E0A9-E50E24DCCA9E"));
                    this.Characteristic_TX = await this.Service_Selected.GetCharacteristicAsync(Guid.Parse("6e400002-b5a3-f393-e0a9-e50e24dcca9e"));
                    this.Characteristic_RX = await this.Service_Selected.GetCharacteristicAsync(Guid.Parse("6e400003-b5a3-f393-e0a9-e50e24dcca9e"));
                    await Task.Delay(1000);

                    this.Characteristic_RX.ValueUpdated += (o, args) =>
                    {
                        var bytes = args.Characteristic.Value;
                    };
                    await Task.Delay(1000);
                    await this.Characteristic_RX.StartUpdatesAsync();
                    await Task.Delay(1000);

                    byte[] octet = new byte[1];
                    octet[0] = 0x82;
                    var y = await this.Characteristic_TX.WriteAsync(octet);
                }));
                
            };
```

## Expected behavior
it should go into the event just after writing the TX characteristic

## Actual behavior
It don't go into event.

### Crashlog

none

## Configuration

**Version of the Plugin:** 1.3.0

**Platform:** e.g.  Android 8.0

**Device:** Huawei P20 light

suriyaKalidoss21, 2019-09-18:
Hi @Joden31 
 is this issue fixed????

LinuxAndOracle, 2019-10-21:
The following passage was translated by me. I read all the questions about ValueUpdated, but I didn't have a satisfactory result. I really have no way. I even use Google Translate to express what I want to say: How can I get a reply to a Bluetooth device, my code is almost the same as @Joden31 
如果可以得到大家的帮助，我将万分感谢！

ilakidain, 2020-02-17:
Hi @Joden31
I had the same problem.
 The problem was in the size of the MTU (I don't know what the default size is)
_(In my case the length of bytes returned was 27.)_

The solution:   **var j = _pumpSelected.Device.RequestMtuAsync(32);**

---------------
Device.BeginInvokeOnMainThread(async () =>
		{
			try
			{
				var j = _pumpSelected.Device.RequestMtuAsync(32);

				ServiceConfiguracion = await _pumpSelected.Device.GetServiceAsync(guidService_Configuration);
				characteristic_Config_Writeable = await ServiceConfiguracion.GetCharacteristicAsync(guidCharacterictic_Config_Writeable);
				characteristic_Config_Notification = await ServiceConfiguracion.GetCharacteristicAsync(guidCharacterictic_Config_Notification);
			
				await characteristic_Config_Notification.StartUpdatesAsync();

				characteristic_Config_Notification.ValueUpdated -= Characteristic_Config_Notification_ValueUpdated;
				characteristic_Config_Notification.ValueUpdated += Characteristic_Config_Notification_ValueUpdated;

			}
			catch (Exception ex )
			{
				_userDialogs.Alert(ex.Message, "TestCommandCONNECT", "Ok");
				
			}

		});
--------------
I hope it's helpful to you

ffrige, 2020-07-11:
@Joden31 @xabre were you guys able to find what causes this issue?
I have the same problem here: I can write and read fine but I cannot get notifications. The code where I call StartUpdatesAsync does not generate any exception, but the ValueUpdated event is never called.

I tried with a 3rd party application on the same Android device and it works fine, so the server is sending out data correctly, I just can't receive it on my side.

I also tried the suggestion mentioned by @ilakidain but it did not change anything, unfortunately.

What else could it be? Permissions? Timings?
Thanks for your help.

ffrige, 2020-07-16:
Since no activity or response was given here I decided to switch to another similar Bluetooth package (https://github.com/nexussays/ble.net) and found that it works flawlessly.