dotnet-bluetooth-le/dotnet-bluetooth-le #356: BLE in Windows Application?

jvhgamer, 2019-04-24:
Following the readme, under header "## Usage" we are stuck at initializing the plug-in.

So-
How to address portable version of this assembly in a Windows application?
or/and
Where to initialize Mvx.Resolve in a Windows Application?

## Steps to reproduce

1. Add Plugin.BLE / MvvmCross.Plugin.BLE via NuGet to Windows Application VS solution.

2. Attempt to initialize with either the 'Vanilla' or 'MvvmCross' calls to init BLE.

3. Error(s) will produce

## Expected behavior
Tell us what should happen
Run

## Actual behavior
Tell us what happens instead
Errors out

### Crashlog

If something causes an exception paste full stack trace + Exception here

```
//**Vanilla**
try
{
    var ble = CrossBluetoothLE.Current;
    var adapter = CrossBluetoothLE.Current.Adapter;
}
catch (Exception ex)
{
    Console.WriteLine(ex.Message);
}
```

  | Name | Value | Type
-- | -- | -- | --
  | Message | "This functionality is not implemented in the portable version of this assembly.  You should reference the NuGet package from your main application project in order to reference the platform-specific implementation." | string

  | Name | Value | Type
-- | -- | -- | --
  | StackTrace | "   at Plugin.BLE.CrossBluetoothLE.get_Current()\r\n   at MyWindowsApp.Source.Logic.InitBLE() in C:\\...\\MyWindowsApp\\Source\\Logic.cs:line 116" | string

```
//**MvvmCross**
try
{
    var ble = Mvx.Resolve<IBluetoothLE>();
    var adapter = Mvx.Resolve<IAdapter>();
}
catch (Exception ex)
{
    Console.WriteLine(ex.Message);
}
```

  | Name | Value | Type
-- | -- | -- | --
  | Message | "Object reference not set to an instance of an object." | string

  | Name | Value | Type
-- | -- | -- | --
  | StackTrace | "   at MvvmCross.Platform.Mvx.Resolve[TService]()\r\n   at MyWindowsApp.Source.Logic.InitBLE() in C:\\...\\MyWindowsApp\\Source\\Logic.cs:line 127" | string

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
1.3.0

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
Windows Application using .NET Framework (targets 4.5.1)

**Device:** e.g. HTC Sensation /i Phone 7 ...
PC running Windows 10

piersdeseilligny, 2019-11-25:
Any update on this? Does this library actually even work with UWP at all?

xabre, 2019-11-25:
@piersdeseilligny try the latest pre release and try the sample app

VictorKochetkov, 2020-01-17:
Same problem have faced on UWP :(

naryk86, 2022-09-26:
It happens the also using the plugin with .net Maui: same error when trying to start the project on "Windows Machine". Any idea?

emanuelevaini, 2022-11-16:
Thanks @janusw!
I downloaded the nupkg you told me and what happens is the following.
I shoud say firstly that I'm trying to use the plugin on net maui and the issue not occurred debugging on Android.

Now in Windows Machine I can successfully instantiate
 ble = CrossBluetoothLE.Current;
 adapter = CrossBluetoothLE.Current.Adapter;

but when I start searching for devices I have the following message
System.NullReferenceException: 'Object reference not set to an instance of an object.'
which in the call stack seems to be related to 
CommunityToolkit.WinUI.Connectivity.dll!CommunityToolkit.WinUI.Connectivity.ObservableBluetoothLEDevice.LoadGlyph()
because it said "No symbol loaded" (or similar, in english)

I then tried to Disable "Just My Code" (as VS2022 suggested) but seems that nothing changed,

Do you have any idea?

Anyway @janusw , I hope that my answer helps you in resolving bug issues... I am a beginner in plugin testing...

janusw, 2022-11-16:
Thanks for the feedback, @emanuelevaini! Very helpful!

> Now in Windows Machine I can successfully instantiate ble = CrossBluetoothLE.Current; adapter = CrossBluetoothLE.Current.Adapter;

Good, since this was not working previously (IIUC), it seems we're making some progress at least! 👍 

> but when I start searching for devices I have the following message System.NullReferenceException: 'Object reference not set to an instance of an object.' which in the call stack seems to be related [...]

This seems to be yet another problem specific to MAUI on Windows. One would need to look into this. Unfortunately I'm currently not well equipped for debugging on Windows ...

janusw, 2022-11-17:
> but when I start searching for devices I have the following message System.NullReferenceException: 'Object reference not set to an instance of an object.' which in the call stack seems to be related [...]

Ok, I can actually reproduce this problem, using this MAUI app: https://github.com/TimoKinnunen/MauiXAMLBluetoothLE

If I run this on Windows as is, there is no error (and scanning finds no devices), but if I apply the following modification (in order to scan for all possible devices) ...

```
diff --git a/MauiXAMLBluetoothLE/Services/BluetoothLEService.cs b/MauiXAMLBluetoothLE/Services/BluetoothLEService.cs
index 65131ba..dc0f36a 100644
--- a/MauiXAMLBluetoothLE/Services/BluetoothLEService.cs
+++ b/MauiXAMLBluetoothLE/Services/BluetoothLEService.cs
@@ -42,7 +42,7 @@ public class BluetoothLEService
                     await ShowToastAsync($"Found {systemDevice.State.ToString().ToLower()} device {systemDevice.Name}.");
                 }
             }
-            await Adapter.StartScanningForDevicesAsync(HeartRateUuids.HeartRateServiceUuids);
+            await Adapter.StartScanningForDevicesAsync();
         }
         catch (Exception ex)
         {
```

... then I get the same NullReferenceException that you saw:

```
System.NullReferenceException
  HResult=0x80004003
  Message=Object reference not set to an instance of an object.
  Source=CommunityToolkit.WinUI
  StackTrace:
   at CommunityToolkit.WinUI.DispatcherQueueExtensions.EnqueueAsync(DispatcherQueue dispatcher, Func`1 function, DispatcherQueuePriority priority)
   at CommunityToolkit.WinUI.Connectivity.ObservableBluetoothLEDevice.<LoadGlyph>d__64.MoveNext()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
   at System.Threading.Thread.StartCallback()
```

This happens also with the latest master version (2.2.0-pre5-34-g17221bc).

janusw, 2022-11-17:
However, using the BLE.Client.UWP sample app, scanning (also without filter) seems to work fine and returns some results.

So it seems this is a MAUI-only problem on Windows (while Xamarin.UWP works well).

emanuelevaini, 2022-11-17:
Thanks @Janus,
is there something you suggest me to do, or -better- is there something that we could (try) to do to help ?

janusw, 2022-11-17:
> is there something you suggest me to do, or -better- is there something that we could (try) to do to help ?

Not much right now. I already have a suspicion what might be going wrong (and how to fix it).

janusw, 2022-11-17:
In particular my suspicion is that our usage of CommunityToolkit.WinUI.Connectivity v7.x in net6.0-windows is the problem. For Xamarin.UWP, we did not update Microsoft.Toolkit.Uwp.Connectivity from v6.x to v7.x yet, because the update caused problems. I assume these are the same problems we see here.

The easy fix would be to downgrade CommunityToolkit.WinUI.Connectivity from 7.1.2 to 6.x, but unfortunately nuget.org only seems to have 7.x versions :/

The proper fix would be to dive into the underlying problem and update our usage of CommunityToolkit.WinUI.Connectivity so that we can use v7. I'm not sure if I can get to that anytime soon, though.

@emanuelevaini If you feel like you could look into any of this: Any help is certainly appreciated!

fomlhaut, 2022-11-21:
> at CommunityToolkit.WinUI.Connectivity.ObservableBluetoothLEDevice.<LoadGlyph>d__64.MoveNext()

According to the stack trace, the error is being thrown during a call in one of the extension methods EnqueueAsync:
[Microsoft.Toolkit.Uwp/Extensions/DispatcherQueueExtensions.cs#L34](https://github.com/CommunityToolkit/WindowsCommunityToolkit/blob/77b009ddf591b78dfc5bad0088c99ce35406170b/Microsoft.Toolkit.Uwp/Extensions/DispatcherQueueExtensions.cs#L34)

Those extension methods would throw a `NullReferenceException` if the `this DispatcherQueue dispatcher` parameter was null when checking if `dispatcher.HasThreadAccess` is true in the first `if`.

If we have a look at the calling function according to the stack trace, which is this one here:
[Microsoft.Toolkit.Uwp.Connectivity/BluetoothLEHelper/ObservableBluetoothLEDevice.cs#L545](https://github.com/CommunityToolkit/WindowsCommunityToolkit/blob/77b009ddf591b78dfc5bad0088c99ce35406170b/Microsoft.Toolkit.Uwp.Connectivity/BluetoothLEHelper/ObservableBluetoothLEDevice.cs#L545)
```
        /// <summary>
        /// Load the glyph for this device
        /// </summary>
        private async void LoadGlyph()
        {
            await DispatcherQueue.EnqueueAsync(
                async () =>
                {
                    var deviceThumbnail = await DeviceInfo.GetGlyphThumbnailAsync();
                    var glyphBitmapImage = new BitmapImage();
                    await glyphBitmapImage.SetSourceAsync(deviceThumbnail);
                    Glyph = glyphBitmapImage;
                }, DispatcherQueuePriority.Normal);
        }
```
We see that it gets called from the ObservableBluetoothLEDevice constructor here:
[Microsoft.Toolkit.Uwp.Connectivity/BluetoothLEHelper/ObservableBluetoothLEDevice.cs#L146](https://github.com/CommunityToolkit/WindowsCommunityToolkit/blob/77b009ddf591b78dfc5bad0088c99ce35406170b/Microsoft.Toolkit.Uwp.Connectivity/BluetoothLEHelper/ObservableBluetoothLEDevice.cs#L146)

There, if the `dispatcherQueue` parameter is null, it will be assigned the value returned by `DispatcherQueue.GetForCurrentThread()`, which, according to the documentation, can return null...

Apart from the obvious fact that there are no null-checks, the function `DispatcherQueue.GetForCurrentThread()` returns a DispatcherQueue only if it's called on a thread that actually has a DispatcherQueue. So, if it's called on a background process thread, then it will be null.

The answer [here](https://stackoverflow.com/questions/69873333/dispatcherqueue-null-when-trying-to-update-ui-property-in-viewmodel) proposes to call the method on the UI thread and then store the return value in a variable that can be used from the background thread.

It should be fixed in the project [https://github.com/CommunityToolkit/WindowsCommunityToolkit](https://github.com/CommunityToolkit/WindowsCommunityToolkit), shouldn't it?

fomlhaut, 2022-11-22:
I'd like to add what I think the failing call sequence is:

1. In Plugin.BLE, `StartScanningForDevicesAsync()` calls `StartScanningForDevicesNativeAsync()`, which in turn sets `DeviceFoundAsync()` as the `Received` event handler of the BluetoothLEAdvertisementWatcher, and then calls `_bleWatcher.Start()`.
3. When the `DeviceFoundAsync()` event handler runs, it calls `new Device()`.
4. The `Device()` constructor calls the `new ObservableBluetoothLEDevice()` constructor (from CommunityToolkit.WinUI).
5. Here the call to `LoadGlyph()` would throw an error if `DispatcherQueue` is null.

What do you think?
If all this makes any sense, why would the `DeviceFoundAsync()` event handler run in a non-UI thread?

janusw, 2022-11-23:
@emanuelevaini @fomlhaut I created PR #640, which is supposed to fix this problem. Testing and feedback welcome!

emanuelevaini, 2022-11-23:
Thanks @janusw !
I want to try it but... how? Have I to get a new nuget pck?
Sorry for the noob question...

janusw, 2022-11-23:
The nupkg is at https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/suites/9470243187/artifacts/448537887 (built by the GitHub Actions CI).

fomlhaut, 2022-11-24:
> @emanuelevaini @fomlhaut I created PR #640, which is supposed to fix this problem. Testing and feedback welcome!

Thank you very much for your help! I've seen that you have already fixed it and published the 3.0.0-beta.2. I will try it as soon as possible and tell you how it goes. I was trying to use it in a UNO Platform app (which uses WinUI 3 instead of UWP) and was getting the error.

janusw, 2022-11-24:
> > @emanuelevaini @fomlhaut I created PR #640, which is supposed to fix this problem. Testing and feedback welcome!
> 
> Thank you very much for your help!

Dito! 😉 (without your analysis of the problem I could not have fixed it so quickly 😃)

> I've seen that you have already fixed it and published the 3.0.0-beta.2. I will try it as soon as possible and tell you how it goes.

Ok, thanks. I'm pretty sure scanning on Windows works well now, both with Xamarin as well as .NET6/MAUI. Therefore I'll close this issue (if you find any further problems, please open a new one).

closed 2022-11-24 by janusw