dotnet-bluetooth-le/dotnet-bluetooth-le #362: How to use this plugin in UWP

wangguan1888, 2019-07-11:
It seems that this plugin not support UWP.
I see a branch which can support UWP. Its name is "uwp_creators_update".
https://github.com/xabre/xamarin-bluetooth-le/branches

I would like to know how to use that branch in my project.

MenaceSan, 2019-07-11:
I've used the UWP branch. It mostly works . I had to make a few minor fixes to it. null checks, etc.
You dont use the whole branch . just pull the UWP project out of it. 
Just copy the UWP project (Plugin.BLE.UWP) and include it as a referenced project (Not using NuGet)
You then need to init it in your UWP startup project like:
var bleReal = new Plugin.BLE.BleImplementation();    // Allow UWP access to real BT device.
            bleReal.Initialize();       // need to call this to make the device work.
Then just use the bleReal in the .NET Standard shared part of your project.
I've seen some devices not work properly that otherwise do in iOS and Android. I have no idea if that is a failing of the BLE code or the underlying Windows stack. Might be either.

As a side note. I ALWAYS build a UWP project in my Xamarin solutions if I can. The debugging capabilities and having Edit and Continue make it worth the trouble. Even if no users users it.

xabre, 2019-08-11:
Preview release available with 2.2.0-pre1

closed 2019-08-11 by xabre

sisaacks, 2021-09-02:
@xabre I am seeing issues with _adapter.DisconnectDeviceAsync().......it just hangs and never returns in UWP.  If I call it without Async it says the device disconnected but the next time I try to discover the device, it says the device is connected.....when it said it was disconnected in the disconnecting process.