dotnet-bluetooth-le/dotnet-bluetooth-le #370: Discovering Advertising Beacons on iOS and Android

HeadBanging, 2019-08-05:
I am trying to use your code to use an advertising beacon - all the code needs to do is register the presence of a device with a particular UUID and, if the device is seen, raise an alert.  Nothing more.  It doesn't need to communicate with the device - but it would be nice if it could log the signal strength and work in the background.

I can get it scan for visible Bluetooth devices but, on iOS at least, the GUIDs that I see don't appear to be genuine.  In the Swift code that I wrote (iOS only) I was able to use centralManager didDiscoverPeripheral to accomplish what I needed but, now that I need to accommodate Android as well this (obviously) doesn't work.

(This is in MainPage.xaml.cs)

    public partial class MainPage : TabbedPage {

        IBluetoothLE ble;
        IAdapter adaptor;
        ObservableCollection<IDevice> deviceList;

        async Task ScanForBLE() {

            while (true) {
                adaptor.DeviceDiscovered += (s, a) => {
                    deviceList.Add(a.Device);
                };
                await adaptor.StartScanningForDevicesAsync();
            }
        }

        public MainPage() {
            InitializeComponent();
            ble = CrossBluetoothLE.Current;
            adaptor = CrossBluetoothLE.Current.Adapter;
            deviceList = new ObservableCollection<IDevice>();
            deviceList.Clear();
            
            _ = ScanForBLE();
            
            this.Children.Add(new ScanPage());
            this.Children.Add(new ConnectPage());
        }
    }

The code above code works - but it doesn't find the GUID of advertising peripherals.  

What I need is a cross platform Xamarin equivalent of the code below:

    func centralManager(_ central: CBCentralManager, didDiscover peripheral: CBPeripheral, advertisementData: [String : Any], rssi RSSI: NSNumber) {
        if let list = advertisementData["kCBAdvDataServiceUUIDs"] as? [AnyObject], (list.contains { ($0 as? CBUUID)?.uuidString == "XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX" }) {
            // Do Stuff Here
        }
    }

To be clear, this isn't a bug.  Xamarin-Bluetooth-LE is working - I just can't get to do what I need on Android and iOS.  Any suggestions on what I might do to accomplish what I need will be gratefully received.

xabre, 2019-08-06:
Take a look at the IAdapter method for scanning. You probably need to set the `serviceUuids optional parameter :)

```CSHARP
        /// <summary>
        /// Starts scanning for BLE devices that fulfill the <paramref name="deviceFilter"/>.
        /// DeviceDiscovered will only be called, if <paramref name="deviceFilter"/> returns <c>true</c> for the discovered device.
        /// </summary>
        /// <param name="serviceUuids">Requested service Ids. The default is null.</param>
        /// <param name="deviceFilter">Function that filters the devices. The default is a function that returns true.</param>
        /// <param name="allowDuplicatesKey"> iOS only: If true, filtering is disabled and a discovery event is generated each time the central receives an advertising packet from the peripheral. 
        /// Disabling this filtering can have an adverse effect on battery life and should be used only if necessary.
        /// If false, multiple discoveries of the same peripheral are coalesced into a single discovery event. 
        /// If the key is not specified, the default value is false.
        /// For android, key is ignored.</param>
        /// <param name="cancellationToken">The token to monitor for cancellation requests. The default value is None.</param>
        /// <returns>A task that represents the asynchronous read operation. The Task will finish after the scan has ended.</returns>
        Task StartScanningForDevicesAsync(Guid[] serviceUuids = null, Func<IDevice, bool> deviceFilter = null, bool allowDuplicatesKey = false, CancellationToken cancellationToken = default);
```

HeadBanging, 2019-08-07:
Thank you for responding so quickly!

I've tried the following:

            while (true) {
                adaptor.DeviceDiscovered += (s, a) => {
                    deviceList.Add(a.Device);
                };
                await adaptor.StartScanningForDevicesAsync(new[] { Guid.Parse("XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX") });
            }

Where XX is the GUID of the BLE beacon that I'm trying to discover.  Even using this code though, it seems to return all beacons - not just the beacon that I'm looking for.

The logic should be quite simple - If I see the beacon that I'm interested in, do something.  Otherwise do nothing.

Being a beacon, I'm not trying to connect to it - but the beacon functionality hasn't been implemented entirely correctly.  The beacon appears as a peripheral - hence, in swift, having to use 'diddiscover peripheral'

xabre, 2019-08-07:
Well don't all your beacons advertise the same **service** UUIDS based on the fact that they are the same device type and from the same manufacturer?

StartScanningForDevicesAsync has two optional parameters: one for advertised service uuids, and a filter for device uuid

HeadBanging, 2019-08-09:
Xabre - my apologies, your code works well.  It was my error!

closed 2019-08-09 by xabre