dotnet-bluetooth-le/dotnet-bluetooth-le #373: ConnectToKnownDeviceAsync Crash When Bluetooth is not in Range in Version 2.1.0 with iOS 12+

BlueFire-LLC, 2019-08-14:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1.

2.

3.

## Expected behavior
Tell us what should happen

## Actual behavior
Tell us what happens instead

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)

**Device:** e.g. HTC Sensation /i Phone 7 ...

BlueFire-LLC, 2019-08-14:
Just discovered this and going back to 2.0.1 fixed the issue. Hard to provide a test case as it's buried deep into my App. No problem with Android, only iOS 12+. Ok with iOS 9+. Not sure about 10 or 11. Hopefully it is something easily fixed but let me know if I can provide anything else.

xabre, 2019-08-28:
Should be fixed by release 2.1.1.

closed 2019-08-28 by xabre

BlueFire-LLC, 2019-08-30:
2.1.1 fixed it. Thanks.