dotnet-bluetooth-le/dotnet-bluetooth-le #375: App crashes when connected device is turned off (Android Only)

MarcAlx, 2019-08-19:
When a connected device is turned off, the app doesn't fire `CrossBluetoothLE.Current.Adapter.DeviceConnectionLost` event, app crashes instead.

On iOS the event is fired, seems related to the Android implementation of the plugin.

Do not hesitate to ask for precision. 

Thanks for your help.

## Steps to reproduce

1. Start discovering bluetooth device
2. Once a device is found, stop discovery and connect
3. Read/write some characteristics
4. Turn off device
5. Wait for seconds or try to access device.
6. The app crashes

## Expected behavior
The app should not crashes and the `CrossBluetoothLE.Current.Adapter.DeviceConnectionLost` event should be fired.

## Actual behavior
The app crashes without any possibility to catch the thrown exception.

### Crashlog

```
[MonoDroid] UNHANDLED EXCEPTION:
[MonoDroid] System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> Java.Lang.RuntimeException: Can't create handler inside thread that has not called Looper.prepare()
[MonoDroid]   at Java.Interop.JniEnvironment+InstanceMethods.CallObjectMethod (Java.Interop.JniObjectReference instance, Java.Interop.JniMethodInfo method, Java.Interop.JniArgumentValue* args) [0x00069] in <aaca87efbce04e508c2ab386cb0cb51a>:0 
[MonoDroid]   at Java.Interop.JniPeerMembers+JniInstanceMethods.InvokeVirtualObjectMethod (System.String encodedMember, Java.Interop.IJavaPeerable self, Java.Interop.JniArgumentValue* parameters) [0x0002a] in <aaca87efbce04e508c2ab386cb0cb51a>:0 
[MonoDroid]   at Android.App.AlertDialog+Builder.Create () [0x0000a] in <60cd5c1750264e4eaf5076c954fcf292>:0 
[MonoDroid]   at Xamarin.Forms.Platform.Android.PopupManager+PopupRequestHelper.OnAlertRequested (Xamarin.Forms.Page sender, Xamarin.Forms.Internals.AlertArguments arguments) [0x00022] in <3bb4fba96f574e50b416a64c2877f966>:0 
[MonoDroid]   at (wrapper managed-to-native) System.Reflection.RuntimeMethodInfo.InternalInvoke(System.Reflection.RuntimeMethodInfo,object,object[],System.Exception&)
[MonoDroid]   at System.Reflection.RuntimeMethodInfo.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0006a] in <a9f928dd2a284728a87c3372b33e3685>:0 
[MonoDroid]    --- End of inner exception stack trace ---
[MonoDroid]   at System.Reflection.RuntimeMethodInfo.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00086] in <a9f928dd2a284728a87c3372b33e3685>:0 
[MonoDroid]   at System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) [0x00000] in <a9f928dd2a284728a87c3372b33e3685>:0 
[MonoDroid]   at Xamarin.Forms.MessagingCenter+Subscription.InvokeCallback (System.Object sender, System.Object args) [0x00076] in <fbf57c0c5c3c4c82ba74be6c0c1f08d0>:0 
[MonoDroid]   at Xamarin.Forms.MessagingCenter.InnerSend (System.String message, System.Type senderType, System.Type argType, System.Object sender, System.Object args) [0x0006b] in <fbf57c0c5c3c4c82ba74be6c0c1f08d0>:0 
[MonoDroid]   at Xamarin.Forms.MessagingCenter.Xamarin.Forms.IMessagingCenter.Send[TSender,TArgs] (TSender sender, System.String message, TArgs args) [0x00013] in <fbf57c0c5c3c4c82ba74be6c0c1f08d0>:0 
[MonoDroid]   at Xamarin.Forms.MessagingCenter.Send[TSender,TArgs] (TSender sender, System.String message, TArgs args) [0x00005] in <fbf57c0c5c3c4c82ba74be6c0c1f08d0>:0 
[MonoDroid]   at Xamarin.Forms.Page.DisplayAlert (System.String title, System.String message, System.String accept, System.String cancel) [0x0001f] in <fbf57c0c5c3c4c82ba74be6c0c1f08d0>:0 
[MonoDroid]   at Xamarin.Forms.Page.DisplayAlert (System.String title, System.String message, System.String cancel) [0x00000] in <fbf57c0c5c3c4c82ba74be6c0c1f08d0>:0 
[MonoDroid]   at MYAPP.MyEventHandler (System.Object sender, System.Object e) [0x0002c] in <8a5a28a9cf6542b2aaded87d88453c01>:0 
[MonoDroid]   at MYAPP.MyMethod.<Init>b__52_2 (System.Object sender, Plugin.BLE.Abstractions.EventArgs.DeviceErrorEventArgs e) [0x0000a] in <8a5a28a9cf6542b2aaded87d88453c01>:0 
[MonoDroid]   at Plugin.BLE.Abstractions.AdapterBase.HandleDisconnectedDevice (System.Boolean disconnectRequested, Plugin.BLE.Abstractions.Contracts.IDevice device) [0x00065] in <12956cfcb4ff491aa0cc1f7ab64b7e72>:0 
[MonoDroid]   at Plugin.BLE.Android.GattCallback.OnConnectionStateChange (Android.Bluetooth.BluetoothGatt gatt, Android.Bluetooth.GattStatus status, Android.Bluetooth.ProfileState newState) [0x001d7] in <2ac92a23dc8c491ca71069bfaf396e20>:0 
[MonoDroid]   at Android.Bluetooth.BluetoothGattCallback.n_OnConnectionStateChange_Landroid_bluetooth_BluetoothGatt_II (System.IntPtr jnienv, System.IntPtr native__this, System.IntPtr native_gatt, System.Int32 native_status, System.Int32 native_newState) [0x00016] in <60cd5c1750264e4eaf5076c954fcf292>:0 
[MonoDroid]   at (wrapper dynamic-method) Android.Runtime.DynamicMethodNameCounter.96(intptr,intptr,intptr,int,int)
[art] JNI RegisterNativeMethods: attempt to register 0 native methods for android.runtime.JavaProxyThrowable
[JavaBinder] *** Uncaught remote exception!  (Exceptions are not yet supported across processes.)
[JavaBinder] android.runtime.JavaProxyThrowable: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> Java.Lang.RuntimeException: Can't create handler inside thread that has not called Looper.prepare()
[JavaBinder]   at Java.Interop.JniEnvironment+InstanceMethods.CallObjectMethod (Java.Interop.JniObjectReference instance, Java.Interop.JniMethodInfo method, Java.Interop.JniArgumentValue* args) [0x00069] in <aaca87efbce04e508c2ab386cb0cb51a>:0 
[JavaBinder]   at Java.Interop.JniPeerMembers+JniInstanceMethods.InvokeVirtualObjectMethod (System.String encodedMember, Java.Interop.IJavaPeerable self, Java.Interop.JniArgumentValue* parameters) [0x0002a] in <aaca87efbce04e508c2ab386cb0cb51a>:0 
[JavaBinder]   at Android.App.AlertDialog+Builder.Create () [0x0000a] in <60cd5c1750264e4eaf5076c954fcf292>:0 
[JavaBinder]   at Xamarin.Forms.Platform.Android.PopupManager+PopupRequestHelper.OnAlertRequested (Xamarin.Forms.Page sender, Xamarin.Forms.Internals.AlertArguments arguments) [0x00022] in <3bb4fba96f574e50b416a64c2877f966>:0 
[JavaBinder]   at (wrapper managed-to-native) System.Reflection.RuntimeMethodInfo.InternalInvoke(System.Reflection.RuntimeMethodInfo,object,object[],System.Exception&)
[JavaBinder]   at System.Reflection.RuntimeMethodInfo.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0006a] in <a9f928dd2a284728a87c3372b33e3685>:0 
[JavaBinder]    --- End of inner exception stack trace ---
[JavaBinder]   at System.Reflection.RuntimeMethodInfo.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00086] in <a9f928dd2a284728a87c3372b33e3685>:0 
[JavaBinder]   at System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) [0x00000] in <a9f928dd2a284728a87c3372b33e3685>:0 
[JavaBinder]   at Xamarin.Forms.MessagingCenter+Subscription.InvokeCallback (System.Object sender, System.Object args) [0x00076] in <fbf57c0c5c3c4c82ba74be6c0c1f08d0>:0 
[JavaBinder]   at Xamarin.Forms.MessagingCenter.InnerSend (System.String message, System.Type senderType, System.Type argType, System.Object sender, System.Object args) [0x0006b] in <fbf57c0c5c3c4c82ba74be6c0c1f08d0>:0 
[JavaBinder]   at Xamarin.Forms.MessagingCenter.Xamarin.Forms.IMessagingCenter.Send[TSender,TArgs] (TSender sender, System.String message, TArgs args) [0x00013] in <fbf57c0c5c3c4c82ba74be6c0c1f08d0>:0 
[JavaBinder]   at Xamarin.Forms.MessagingCenter.Send[TSender,TArgs] (TSender sender, System.String message, TArgs args) [0x00005] in <fbf57c0c5c3c4c82ba74be6c0c1f08d0>:0 
[JavaBinder]   at Xamarin.Forms.Page.DisplayAlert (System.String title, System.String message, System.String accept, System.String cancel) [0x0001f] in <fbf57c0c5c3c4c82ba74be6c0c1f08d0>:0 
[JavaBinder]   at Xamarin.Forms.Page.DisplayAlert (System.String title, System.String message, System.String cancel) [0x00000] in <fbf57c0c5c3c4c82ba74be6c0c1f08d0>:0 
[... cut ...]
[System.err] 	at md5b8da2e4cd7b57e43a9b1e62f6e4aa245.GattCallback.onConnectionStateChange(GattCallback.java:38)
[System.err] 	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:228)
[System.err] 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
[System.err] 	at android.os.Binder.execTransact(Binder.java:573)
[JavaBinder] Forcefully exiting
```

## Configuration

**Version of the Plugin:** 2.1.0

**Platform:** Reproduced on Android 9 and Android 7

**Device:** Nokia 7 Plus (Android 9) Samsung Galaxy Tab S2 (Android 7)

xabre, 2019-08-19:
@MarcAlx seems to me you are calling `DisplayAlert` directly from your `DeviceConnectionLost` handler. The event handler is not necessarily invoked on the main thread (UI thread) so makes sure do do any ui changes on the UI Thread. with `Device.BeginInvokeOnMainThread(() => yourpage.DisplayAlert..)`.

closed 2019-08-19 by xabre

MarcAlx, 2019-08-20:
@xabre Thanks for your help, your answer help me solve my problem. The android only reproducibility combined with jni exception misleaded me.

smmutlu, 2020-03-25:
Wow was banging my head against the wall for a few hours and finally stumbled upon this post, exact same issue. Thank you @xabre for the beautiful library and your big brain