dotnet-bluetooth-le/dotnet-bluetooth-le #376: [Android] Main thread and serial sync

xabre, 2019-08-20:
Provide a platform specific main thread/serial queue to avoid issues like #276 without the user having to worry about these invokations.

Acceptance Criteria:

- [ ] Can be neabled/disabled by setting (will be enabled by default)
- [ ] A stuck task will not block forever and will time out in a predefined time interval 
- [ ] Extend sample app to test this

xabre, 2019-09-17:
Experimental release 2.2.0-pre2. Would be great to get some feedback.

hudaIT, 2019-09-20:
Should we post feedback directly here?

Umer-Mahmood, 2021-05-03:
Is this part of the plugin or not yet?

Umer-Mahmood, 2021-05-03:
> Experimental release 2.2.0-pre2. Would be great to get some feedback.

It did solve my problem. Thanks alot. Have not had any problem with 2.2.0-pre2 so far. 
I always got "Gatt write discriptor value failed" when trying to do multiple StartUpdatesAsync calls. But not anymore.

RobertLinak, 2022-11-03:
@xabre How do you disable this?