dotnet-bluetooth-le/dotnet-bluetooth-le #377: NullReferenceException in iOS' Adapter DisconnectedPeripheral delegate

Tragetaschen, 2019-08-26:
## Steps to reproduce
Unknown so far, the crash was reported through AppCenter and shows nothing particularly interesting in the stack trace.

I am using ConnectToKnownDeviceAsync, so #373 might be related.

## Expected behavior
Besides the entry point, none of my code is on the stack trace, so I don't expect the NRE

## Actual behavior
An NRE with a rather unhelpful stack trace

### Crashlog

```
Xamarin Exception Stack:
System.NullReferenceException: Object reference not set to an instance of an object
  at Plugin.BLE.iOS.Adapter.<.ctor>b__4_3 (System.Object sender, CoreBluetooth.CBPeripheralErrorEventArgs e) <0x105c6b7c0 + 0x0027c> in <09525911aa764ca5850e00803af5b09a#2c17274255111c87b77bb59871e4a881>:0
  at Plugin.BLE.BleCentralManagerDelegate.DisconnectedPeripheral (CoreBluetooth.CBCentralManager central, CoreBluetooth.CBPeripheral peripheral, Foundation.NSError error) <0x105c68e10 + 0x00067> in <09525911aa764ca5850e00803af5b09a#2c17274255111c87b77bb59871e4a881>:0
  at (wrapper managed-to-native) UIKit.UIApplication.UIApplicationMain(int,string[],intptr,intptr)
  at UIKit.UIApplication.Main (System.String[] args, System.IntPtr principal, System.IntPtr delegate) <0x1053c4a20 + 0x0002b> in <d5b15c87344244df901541b57fe34a1e#2c17274255111c87b77bb59871e4a881>:0
  at UIKit.UIApplication.Main (System.String[] args, System.String principalClassName, System.String delegateClassName) <0x1053c4960 + 0x000ab> in <d5b15c87344244df901541b57fe34a1e#2c17274255111c87b77bb59871e4a881>:0
  at [my]Application.Main (System.String[] args) <0x104800520 + 0x00023> in <b2629e786942473985ce6d4cc41689d0#2c17274255111c87b77bb59871e4a881>:0
```

## Configuration

**Version of the Plugin:** 2.1.0

**Platform:** iOS 12.4

**Device:** iPhone XR

Tragetaschen, 2019-08-28:
I have a better stack trace from capturing it in a debug build:

```
Xamarin Exception Stack:
System.NullReferenceException: Object reference not set to an instance of an object
  at Plugin.BLE.iOS.Adapter.<.ctor>b__4_3 (System.Object sender, CoreBluetooth.CBPeripheralErrorEventArgs e) [0x000c9] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.iOS\Adapter.cs:117
  at Plugin.BLE.BleCentralManagerDelegate.DisconnectedPeripheral (CoreBluetooth.CBCentralManager central, CoreBluetooth.CBPeripheral peripheral, Foundation.NSError error) [0x00000] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.iOS\BleCentralManagerDelegate.cs:120
  at (wrapper managed-to-native) UIKit.UIApplication.UIApplicationMain(int,string[],intptr,intptr)
  at UIKit.UIApplication.Main (System.String[] args, System.IntPtr principal, System.IntPtr delegate) [0x00005] in /Library/Frameworks/Xamarin.iOS.framework/Versions/12.14.0.114/src/Xamarin.iOS/UIKit/UIApplication.cs:86
  at UIKit.UIApplication.Main (System.String[] args, System.String principalClassName, System.String delegateClassName) [0x0000e] in /Library/Frameworks/Xamarin.iOS.framework/Versions/12.14.0.114/src/Xamarin.iOS/UIKit/UIApplication.cs:65
  at […].Main (System.String[] args) [0x00000] in […].Mobile.iOS\Main.cs:11
```

https://github.com/xabre/xamarin-bluetooth-le/blob/2.1.0/Source/Plugin.BLE.iOS/Adapter.cs#L117

If the `stringId` doesn't exist, `TryRemove` sets `foundDevice` to `null` which then NREs in the trace message string

Tragetaschen, 2019-08-28:
Talking to my colleague: This happened pretty much exactly when the cancellation token source for the `adapter.ConnectToKnownDeviceAsync` was cancelled.

xabre, 2019-08-28:
Thanks for the good insight @Tragetaschen. Added a fix in the 2.1.1 release

closed 2019-08-28 by xabre