dotnet-bluetooth-le/dotnet-bluetooth-le #379: Unable to Write and read in plugin BLE

suriyaKalidoss21, 2019-09-07:
I need to get the correct characteristic to read and write in Bluetooth device but among two characteristics, I am not able to predicate how to read the value using which characteristic.

characteristic property type-->write/write response

characteristic property type-> indicate

   var services = await connectedDevice.GetServicesAsync();
            var serviceid = services[0].Id;
            var service = await connectedDevice.GetServiceAsync(serviceid);
            var characteristics = await service.GetCharacteristicsAsync();

            var characteristicread= await service.GetCharacteristicAsync(characteristics[1].Id);
            if (characteristicread.CanUpdate)
            {
                Console.WriteLine('yes');
            }
                var characteristicWrite = await service.GetCharacteristicAsync(characteristics[0].Id);
            byte[] bytes = Encoding.ASCII.GetBytes("LIST_DIR,audio");
            await characteristicWrite.WriteAsync(bytes);
          
            characteristicread.ValueUpdated += (s, ee) =>
                {
                    var result = ee.Characteristic.Value;
                    var stre = System.Text.Encoding.UTF8.GetString(result, 0, result.Length);
                   //value not updated
                };
                await characteristicread.StartUpdatesAsync();

#
[image]
# Steps to reproduce

1.unable to read characteristic after write

## Expected behavior
after write values should be updated in read

## Actual behavior
value not updated 

## Configuration

**Platform:** e.g. iOS 10.1

suriyaKalidoss21, 2019-09-07:
i have edited the comment

closed 2019-09-07 by suriyaKalidoss21

YBENZAKOUR, 2020-08-25:
@suriyaKalidoss21  Hello, please do you found the right method to learn and send message? I need it please 
Thank's