dotnet-bluetooth-le/dotnet-bluetooth-le #380: Objective-C exception thrown. Name: NSInternalInconsistencyException Reason: Client Characteristic Configuration descriptors must be configured using setNotifyValue:forCharacteristic:

suriyaKalidoss21, 2019-09-07:
After Start Update Async I got the above exception

> Note: 

code : 
     var services = await connectedDevice.GetServicesAsync();
            
            var serviceid = services[0].Id;
            var service = await connectedDevice.GetServiceAsync(serviceid);
           
            var characteristics = await service.GetCharacteristicsAsync();

            var characteristicread= await service.GetCharacteristicAsync(characteristics[1].Id);
 
             var characteristicWrite = await service.GetCharacteristicAsync(characteristics[0].Id);

            characteristicread.ValueUpdated += (o, args) =>
            {
                var bytes = args.Characteristic.Value;
            };

            try
            {
                await characteristicread.StartUpdatesAsync();
            }
            catch(Exception ex)
            {
             here
            }
          
## Steps to reproduce

1.run the above code.After  characteristicread.StartUpdatesAsync();

2. I got  Client Characteristic Configuration descriptors must be configured using setNotifyValue:characteristic:

## Expected behavior
it should not crash

## Actual behavior
its gets crash

### Crashlog

Objective-C exception thrown. Name: NSInternalInconsistencyException Reason: Client Characteristic Configuration descriptors must be configured using setNotifyValue:forCharacteristic:

## Configuration

**Version of the Plugin:** latest

**Platform:** e.g. iOS 10 

**Device:** iphonexs

suriyaKalidoss21, 2019-09-10:
somebody please provide me quick workaround

xabre, 2019-09-17:
Was not able to reproduce this with the current sample app and an iphone 7 running ios 12.4

suriyaKalidoss21, 2019-09-17:
Hi @xabre 
  I can't able to read from BLE device after write,Please provide me solution Advance Thanks
My code:
             

            var services = await connectedDevice.GetServicesAsync();
            var serviceid = services[0].Id;
            var service = await connectedDevice.GetServiceAsync(serviceid);
            var characteristics = await service.GetCharacteristicsAsync();
            var characteristicread= await service.GetCharacteristicAsync(characteristics[1].Id);
          var characteristicWrite = await service.GetCharacteristicAsync(characteristics[0].Id);
          if (characteristicread.CanUpdate)
            {
                characteristicread.ValueUpdated += (s, ee) =>
                {
                    Debug.WriteLine("characteristic.ValueUpdated");
                    var bytes = ee.Characteristic.Value;
                    string data = Encoding.ASCII.GetString(bytes);  //no value here
                };
                    await characteristicread.StartUpdatesAsync();
            }
            byte[] bytese = Encoding.ASCII.GetBytes("LIST_DIR,audio");
    --write code
                await characteristicWrite.WriteAsync(bytese);
charecteristcis:
[image]

suriyaKalidoss21, 2019-09-17:
Hi @xabre 
  I have two queries,Issues exist in your GitHub samples
**query 1: Unable to read after charecteresteric write'**
        Step 1: Write the bytes to write characteristic,Refer screenshot
        [image]
        
[image]

        Step 2: Unable to read data after the write you can refer the screenshot
[image]

  
**query 2: Exception thrown at descriptor write page, Refer screenshot**
     while writing the same value(write code Refer above code screenshot) to write discriptor I got the exception
[image]

jkandiko, 2019-09-17:
I'm getting the same error when trying to write 02 00 into a descriptor with the sample app.  iPhone is an iPhone SE running 12.4
[image]

suriyaKalidoss21, 2019-09-18:
Hi @jkandiko 
  could you please share me some code sample you have tried.

jkandiko, 2019-09-18:
@suriyaKalidoss21  I didn't modify the any source code from the sample applications.  My guess is your application is writing values that iOS doesn't like and your code is fine.  Here's the input screen on my device

[image]

suriyaKalidoss123, 2019-09-18:
Hi @jkandiko ,
  Thanks for the information. Does it work fine for you?

vimalaven, 2020-06-07:
Hi @suriyaKalidoss123 ,
Are you able to get this resolved? any code samples will help.

In my situation, the device supports only Write & Indicate. I'm not sure if the plugin supports indicate.

@xabre for the devices that doesn't support read characteristics, how can i retrieve values from the device? Any samples would be really helpful.

Thanks,
Vimala

mjphillips1981, 2020-07-17:
Does anyone have a solution for this?  I am getting the same error on iOS.

mpfj, 2021-08-28:
Apologies for dragging up this old issue but I'm getting the same.

[image]

My code works fine under Android.

I've found the following hint at the "solution" but I've no idea what's involved ... [https://stackoverflow.com/questions/57980689/writing-a-notification-descriptor-value-in-ios-using-bluetooth](url)

Is it possible to add this feature / code fix ?

jkandiko, 2021-08-30:
@mpfj That's actually my question and answer.  I'm not really sure why iOS doesn't let you hand write a characteristic into a different state.  In your iOS project, you'll need to create one or two classes that inherit from CBCentralManagerDelegate (to begin searching for devices, discover services, ) and CBPeripheralDelegate (to discover characteristics) to actually write to the characteristic you are looking for.

mpfj, 2021-08-30:
@jkandiko After looking at the BLE plugin source code, I can see that what I'm trying to do is already done for me by the plugin itself (when StartUpdatesAsync() is called)

I finally realised that my code (and the BLE plugin) is working fine ... the only code I need is:-

```
// open the uart service
svcUART = await App.mSelectedDevice.Device.GetServiceAsync(Guid.Parse("49535343-FE7D-4AE5-8FA9-9FAFD205E455"));

// init "tx" (device output) channel
chrTX_DeviceOut = await svcUART.GetCharacteristicAsync(Guid.Parse("49535343-1E4D-4BD9-BA61-23C647249616"));
chrTX_DeviceOut.ValueUpdated += OnReceiveData;
await chrTX_DeviceOut.StartUpdatesAsync();

// init "rx" (device input) channel
chrRX_DeviceIn = await svcUART.GetCharacteristicAsync(Guid.Parse("49535343-8841-43F4-A8D4-ECBE34729BB3"));
```