dotnet-bluetooth-le/dotnet-bluetooth-le #382: Characteristic.StartUpdatesAsync() throwing error [Android]

asfilo, 2019-09-16:
standalone git repository: [https://github.com/asfilo/xamarin-scanner-test/tree/develop](https://github.com/asfilo/xamarin-scanner-test/tree/develop)

## Steps to reproduce

1. Use UI button to connect to device

2. Use UI button to "Get list of services"

3. Characteristic.StartUpdatesAsync() throws an exception

## Expected behavior
App should be listening for device scans. 

NOTE: This was working fine until the tablet went through a series of updates yesterday. 

## Actual behavior
An exception gets thrown

### Crashlog

Exception: Device '00000000-0000-0000-0000-aaa8ac167d9f' disconnected while writing descriptor with 00002902-0000-1000-8000-00805f9b34fb.

StackTrace: 

  at Plugin.BLE.Abstractions.Utils.TaskBuilder.FromEvent[TReturn,TEventHandler,TRejectHandler] (System.Action execute, System.Func`3[T1,T2,TResult] getCompleteHandler, System.Action`1[T] subscribeComplete, System.Action`1[T] unsubscribeComplete, System.Func`2[T,TResult] getRejectHandler, System.Action`1[T] subscribeReject, System.Action`1[T] unsubscribeReject, System.Threading.CancellationToken token) [0x000d0] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\Utils\TaskBuilder.cs:49 
  at Plugin.BLE.Android.Characteristic.StartUpdatesNativeAsync () [0x00218] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:143 
  at scannertest.Views.DeviceView.btnGetServices_Clicked (System.Object sender, System.EventArgs e) [0x00260] in /Users/andyfilo/Projects/scanner-test/scanner-test/Views/DeviceView.xaml.cs:168 

## Configuration

**Version of the Plugin:** 2.1.1

**Platform:** 
Android 7.0 
Kernel version: 3.18.35
Software version: TB-8304F1_RF01_190805
Build version: TB-8304F1_RF01_190805_ROW

**Device:** 
Tablet: Lenovo TB-8304F1
BLE Scanner: Eyoyo-2877 1D

xabre, 2019-09-17:
Hi, can you give the sample app a quick try with your devices? Check if the same issue occurs?

Jkinley01, 2019-09-26:
Also having this issue only on Android after updating to the latest version

softeip, 2023-08-24:
It also throws the same exception on Android 12. Xiaomi Redmi 10X

andreas-spindler-mw, 2024-04-24:
Same on Samsung S8 (Android 8) and Pixel 4 (Android 12) and Pixel 6 (Android 14) if characteristic encryption is enabled.