dotnet-bluetooth-le/dotnet-bluetooth-le #385: [BUG] Automatic reconnect is not possible

hudaIT, 2019-09-20:
There is an issue when i am connected to a device and afterwards close the app, open again and try to connect again. It is not possible to connect to the device again.

## Steps to reproduce

1. Open the sample app, provided in your repository

2. Scan and connect to a device

3. Close the app

4. Scan for a device and try to connect again

## Expected behavior
Successful connect to device

## Actual behavior
Trying to connect endless

### Crashlog

## Configuration

**Version of the Plugin:** e.g. 2.2.0-pre2

**Platform:** Android 9.0

**Device:** Huawei Mediapad T5 Light

veletron2, 2019-11-07:
I also have reconnect issues - but they appear to related to certain android devices. There are many bug reports online regarding android reconnecting issues for stuff not related to xamarin/this plugin (eg to car handsfree systems). I have a variety of devices running a variety of Android versions some work, some don't. Android 6.0 is a particular issue, while some Samsungs on 9.0 reconnect flawlessly, and others intermittently fail. iOS always reconnects flawlessly. Unfortunately, I have had to wrap this issue in a manual reconnect timer which I run every 30 seconds while I am in my 'reconnecting' state. Impact on battery is minimal, and my app runs as a foreground service on Android.

```
ctor
{
 readonly Timer _reconnectRetryTimer;
_reconnectRetryTimer = new Timer { AutoReset = false,Enabled=false,Interval = 30000};
_reconnectRetryTimer.Elapsed += (s, e) => { RestartReconnectionAttempt(); };
_reconnectRetryTimer.Enabled = false;
}

public void RestartReconnectionAttempt()
{
            //dont about for iOS
            if (Device.RuntimePlatform != Device.Android)
                return;

           //stop the timer - we will restart it again
           _reconnectRetryTimer.Enabled = false;

            //This operation is only performed if we are reconnecting
            if (CurrentProteusConnectionState != ConnectionStatus.Reconnecting)
                 return;

            _bleConnectCancellationToken?.Cancel();

           //nothing to wait on here to see if the old connection attempt really was terminated so we have to wing it.

            ConnectToExistingDevice();
       
}

public void ConnectToExistingProteus()
{
            UpdateProteusConnectionState(BleConnectionState.ProteusConnectionStatus.Reconnecting);
            _adapter.ConnectToKnownDeviceAsync(ProteusUuid, new ConnectParameters(autoConnect: true, forceBleTransport: true), _bleConnectCancellationToken.Token);

            if (Device.RuntimePlatform == Device.Android)
                _reconnectRetryTimer.Enabled = true;
}

```

hudaIT, 2019-11-08:
Yes i agree with the fact that this issue only seems to appear on some android devices.

For example reconnecting works fine for Huawei P30 pro, however it does not work for huawei mediapad t5 light

closed 2020-06-20 by Codelisk