dotnet-bluetooth-le/dotnet-bluetooth-le #387: Plugin.BLE crashes app on iPad iOS 13.1

mspoehr, 2019-09-25:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Reference `CrossBluetoothLE.Current` in any capacity

2. Deploy app to iOS running iOS 13

This occurs with a fresh Xamarin.Forms app built with a single button which runs the following code when clicked:

```cs
bool bleEnabled = CrossBluetoothLE.Current.State == BluetoothState.On

## Expected behavior

Code should execute normally without crashing. 

## Actual behavior

App crashes immediately.

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)

**Device:** e.g. HTC Sensation /i Phone 7 ...

mspoehr, 2019-09-25:
I needed to add both NSBluetoothAlwaysUsageDescription and NSBluetoothPeripheralUsageDescription to my info.plist per https://developer.apple.com/documentation/bundleresources/information_property_list/nsbluetoothalwaysusagedescription since our deployment target is < iOS 13.

Everything is working now, but this information should likely be added to the readme so future people don't get hung up on the same issue.

dannythunder, 2019-09-26:
Thanks for this @mspoehr

xabre, 2019-10-14:
Readme was updated thanks to @smstuebe. Will close this issue.

closed 2019-10-14 by xabre