dotnet-bluetooth-le/dotnet-bluetooth-le #390: macOS Mojave and System.NullReferenceException using Visual Studio 2019

Elimil, 2019-10-14:
Hello,
have now tried multiple different BLE libraries and in all have issues at beginning before get any further. With your library happens following

## Steps to reproduce
MvvmCross:
           var ble = Mvx.Resolve<IBluetoothLE>();
Vanilla:
            var ble = CrossBluetoothLE.Current;

## Expected behavior
register IBluetoothLE

## Actual behavior
MvvmCross:
            System.NullReferenceException has been thrown
            Object reference not set to an instance of an object
Vanilla:
            System.ArgumentNullException has been thrown
            Value cannot be null.
            Parameter name: obj

## Configuration
Plugin: Version 2.1.1 from nuget.org
macOS Mojave with latest updates (10.14.6)
Device MacBook Pro 2018
Building with Visual Studio Community 2019
Target framework Xamarin Mac Modern (Have also tried Full version but same behaviour)
And trying to develop application for macOS and no more code to show as stuck at beginning...
any help would be appreciated :)

xabre, 2019-10-15:
@Elimil have you tried the macOS sample? Does it also happen there?

Elimil, 2019-10-15:
Tried and no luck. Following was found from log:
2019-10-15 21:28:48 [TRACE] (MvvmCross.Logging.MvxLog) App Start
2019-10-15 21:28:48 [TRACE] (MvvmCross.Logging.MvxLog) UpdatedState: PoweredOn
2019-10-15 21:28:48 [TRACE] (MvvmCross.Logging.MvxLog) ViewAppeared DeviceListViewModel
2019-10-15 21:28:48 [TRACE] (MvvmCross.Logging.MvxLog) Failed to retreive system connected devices. Value cannot be null.

/Matti

Elimil, 2019-10-15:
Take a little bit back. it works even get the error in the log. Have to look more into the sample and contact back if have more issues.
Thank you anyways...
/Matti

closed 2019-10-15 by Elimil

Elimil, 2019-10-15:
Seems the problem is with system connected devices that library can't scan those on macOS Mojave. But it is not issue in my project so i can continue with it. One neat feature would be that it could also get what is the distance for the device. then it would be perfect, well at least for my project :)

Elimil, 2019-10-15:
And distance seems to be just mathematics 10 ^ ((Measured Power – RSSI)/(10 * N)) where N is Environmental factor in range 2-4. Little googling helps.. So you library is all I need in my project. Thank you :)