dotnet-bluetooth-le/dotnet-bluetooth-le #394: unable to write or read!!

juCortez, 2019-10-23:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1.using DemoBletooth

2.try to use await mycharacteristic.WriteAsync(c) or  await mycharacteristic.ReadAsync(c). 

3.never finish to run !!!

## Expected behavior
read and write 

## Actual behavior
await forever!!!
code:
 var by = new byte[] { 0x20, 0x20 };          
 var wa = await characterRxTx.WriteAsync(c)

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)

**Device:** e.g. HTC Sensation /i Phone 7 ...

************************************************************************************
i am trying to test readAsync and writeAsync inside this method below!!! 

 private async void DevicesList_OnItemSelected(object sender, SelectedItemChangedEventArgs e)
        {
            device = DevicesList.SelectedItem as IDevice;

            var result = await DisplayAlert("AVISO", "Deseja se conectar a esse dispositivo?", "Conectar", "Cancelar");

            if (!result)
                return;

            //Stop Scanner
            await adapter.StopScanningForDevicesAsync();

            try
            {
                await adapter.ConnectToDeviceAsync(device);

                await DisplayAlert("Conectado", "Status:" + device.State, "OK");

            }
            catch (DeviceConnectionException ex)
            {
                await DisplayAlert("Erro", ex.Message, "OK");
            }
            
            var serviceRxTx = await device.GetServiceAsync(Guid.Parse("0000ffe0-0000-1000-8000-00805f9b34fb"));
            var characteristics = await serviceRxTx.GetCharacteristicsAsync();
            var characterRxTx = await serviceRxTx.GetCharacteristicAsync(Guid.Parse("0000ffe1-0000-1000-8000-00805f9b34fb"));
           
            var cw = characteristics[0].CanWrite;
            var cr = characteristics[0].CanRead; 
            var uu = characteristics[0].Uuid;
            var by = new byte[] { 0x20, 0x20 };
            
            var wa = await characterRxTx.WriteAsync(c);
            var ra = await characterRxTx.ReadAsync();
        }

AztiIluna, 2020-03-30:
I am having the same problem. "await characteristic.ReadAsync()" takes forever.
@juCortez Was you able to solve it?

kvrscream, 2020-09-29:
I have a similar problem.
In you're case. Do you need run characterisct.StartUpdateAsync() for indicate to device bluetooth.
But in my android 10 dons't not work. On run await  characterisct.StartUpdateAsync() my code jumping break points and don't catch exeption. On iOS and Android 8.0 run normal.

`  carac.ValueUpdated += (o, args) =>
                                {
                                    var bytes = args.Characteristic.Value;
                                };

                                await carac.StartUpdatesAsync();
`