dotnet-bluetooth-le/dotnet-bluetooth-le #396: on iOS 13, GetServiceAsync() correctly works only if called after GetServicesAsync() ?

roiko, 2019-10-29:
On iphone SE, something goes wrong when I connect to a bluetooth device and try to retrieve a known service, in order to read a characteristic.

## Steps to reproduce

1.  After device discovered, connect to it on main thread             `Device.BeginInvokeOnMainThread(()=>CrossBluetoothLE.Current.Adapter.ConnectToDeviceAsync(StaticData.CurrentDevice, cp));`

2. Retrieve specific service:
`IService service = await StaticData.CurrentDevice.GetServiceAsync(Guid.Parse(serviceUUID));
`
## Expected behavior
Service data is returned

## Actual behavior
An exception is raised and no service can be retrieved. 

The same line of code correctly works if, before executing it, I call GetServicesAsync(). So, the working code is:

```
var servs = await StaticData.CurrentDevice.GetServicesAsync();
IService service = await StaticData.CurrentDevice.GetServiceAsync(Guid.Parse(serviceUUID));

```
### Crashlog

```
System.ArgumentNullException: Value cannot be null.
Parameter name: source
  at System.Linq.Enumerable.TryGetFirst[TSource] (System.Collections.Generic.IEnumerable`1[T] source, System.Func`2[T,TResult] predicate, System.Boolean& found) [0x0000d] in <69f79cbb9a4c4409847837a879ee8812>:0 
  at System.Linq.Enumerable.FirstOrDefault[TSource] (System.Collections.Generic.IEnumerable`1[T] source, System.Func`2[T,TResult] predicate) [0x00000] in <69f79cbb9a4c4409847837a879ee8812>:0 
  at Plugin.BLE.iOS.Device.GetServiceNativeAsync (System.Guid id) [0x00041] in <680562e23cb3427d811efbb788ea743a>:0 
  at Plugin.BLE.Abstractions.DeviceBase`1[TNativeDevice].GetServiceAsync (System.Guid id, System.Threading.CancellationToken cancellationToken) [0x00078] in <ac1ce5528dad47caabfae66238ebe508>:0 
  at VoiceControl.Providers.Bluetooth.BTProvider.ReadLongAttribute (System.String serviceUUID, System.String attributeUUID) [0x0007c] 
```
## Configuration

**Version of the Plugin:** Plugin.BLE 2.2.0-pre2

**Platform:**  iOS 13.1.2

**Device:** iPhone SE

minno726, 2019-12-02:
I am also encountering this error. Here's the stack trace:

```
System.ArgumentNullException: Value cannot be null.
Parameter name: source
  at System.Linq.Enumerable.TryGetFirst[TSource] (System.Collections.Generic.IEnumerable`1[T] source, System.Func`2[T,TResult] predicate, System.Boolean& found) [0x0000d] in <874baba848df4c068fe8cfc0052823f2>:0
  at System.Linq.Enumerable.FirstOrDefault[TSource] (System.Collections.Generic.IEnumerable`1[T] source, System.Func`2[T,TResult] predicate) [0x00000] in <874baba848df4c068fe8cfc0052823f2>:0
  at Plugin.BLE.iOS.Device.GetServiceNativeAsync (System.Guid id) [0x00041] in <680562e23cb3427d811efbb788ea743a>:0
  at Plugin.BLE.Abstractions.DeviceBase`1[TNativeDevice].GetServiceAsync (System.Guid id, System.Threading.CancellationToken cancellationToken) [0x00078] in <ac1ce5528dad47caabfae66238ebe508>:0
  at HvacDemo.Models.MeasurementDevice.ConnectTo (Plugin.BLE.Abstractions.Contracts.IDevice discoveredGauge) [0x001e7] in <snip>
  at HvacDemo.Models.MeasurementDevice.Adapter_DeviceDiscovered (System.Object sender, Plugin.BLE.Abstractions.EventArgs.DeviceEventArgs e) [0x00040] in <snip>
  at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__7_1 (System.Object state) [0x00000] in /Users/builder/jenkins/workspace/xamarin-macios/xamarin-macios/external/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/AsyncMethodBuilder.cs:1037
  at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context (System.Object state) [0x0000d] in /Users/builder/jenkins/workspace/xamarin-macios/xamarin-macios/external/mono/mcs/class/referencesource/mscorlib/system/threading/threadpool.cs:1370
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in /Users/builder/jenkins/workspace/xamarin-macios/xamarin-macios/external/mono/mcs/class/referencesource/mscorlib/system/threading/executioncontext.cs:968
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in /Users/builder/jenkins/workspace/xamarin-macios/xamarin-macios/external/mono/mcs/class/referencesource/mscorlib/system/threading/executioncontext.cs:910
  at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00021] in /Users/builder/jenkins/workspace/xamarin-macios/xamarin-macios/external/mono/mcs/class/referencesource/mscorlib/system/threading/threadpool.cs:1341
  at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in /Users/builder/jenkins/workspace/xamarin-macios/xamarin-macios/external/mono/mcs/class/referencesource/mscorlib/system/threading/threadpool.cs:899
  at ObjCRuntime.Runtime.ThreadPoolDispatcher (System.Func`1[TResult] callback) [0x00006] in /Library/Frameworks/Xamarin.iOS.framework/Versions/13.6.0.12/src/Xamarin.iOS/ObjCRuntime/Runtime.cs:288
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00009] in /Users/builder/jenkins/workspace/xamarin-macios/xamarin-macios/external/mono/mcs/class/referencesource/mscorlib/system/threading/threadpool.cs:1258
```

The line it crashes on is `gaugeService = await gauge.GetServiceAsync(new Guid(...))`, and the crash does not occur if I add the line `var _services = await gauge.GetServicesAsync()`, but don't use the result anywhere.

I'm using Plugin.BLe 2.2.0-pre2, a 7th generation iPad running version 13.1.3, and Xamarin.Forms version 4.3.0.991211.

georgemichailou, 2020-03-19:
I have the same error on ios only. I'm using Plugin.BLe 2.2.0-pre2 and Xamarin.Forms version 4.5.0.396

{System.ArgumentNullException: Value cannot be null.
Parameter name: source
  at System.Linq.Enumerable.TryGetFirst[TSource] (System.Collections.Generic.IEnumerable`1[T] source, System.Func`2[T,TResult] predicate, System.Boolean& found) [0x0000d] in <e7613662fe134bc398f49933939c204b>:0 
  at System.Linq.Enumerable.FirstOrDefault[TSource] (System.Collections.Generic.IEnumerable`1[T] source, System.Func`2[T,TResult] predicate) [0x00000] in <e7613662fe134bc398f49933939c204b>:0 
  at Plugin.BLE.iOS.Device.GetServiceNativeAsync (System.Guid id) [0x00041] in <680562e23cb3427d811efbb788ea743a>:0 
  at Plugin.BLE.Abstractions.DeviceBase`1[TNativeDevice].GetServiceAsync (System.Guid id, System.Threading.CancellationToken cancellationToken) [0x00078] in <ac1ce5528dad47caabfae66238ebe508>:0 
  at Smartin.MainPage.DevicesList_OnItemSelected (System.Object sender, Xamarin.Forms.SelectedItemChangedEventArgs e) [0x00299] in /Users/marinosaimoniotis/Desktop/Smartin/Smartin/MainPage.xaml.cs:131 }

What is the solution for this ?

Abesoddy, 2020-11-05:
Same error, any solution?

commit 833fb008 (2019-12-19): fixed #396 (#402)

GetServiceAsync() correctly works only if called after GetServicesAsync() on iOS 13
closed 2022-06-11 by janusw