dotnet-bluetooth-le/dotnet-bluetooth-le #397: Turn On/Off Bluetooth

bnserig, 2019-11-06:
Hi,

I'm looking for a function to turn on or off bluetooth. I didn't find something like that in CrossBluetoothLE.Current. I have the State (On/Off/TurningOn/TurningOff) but I can't control it.
Is this functionnality planned ?

Thanks.

veletron2, 2019-11-07:
I am pretty sure that this is impossible in iOS - the only way to do this is with the manufacturers UI. You can do it in Android, however, just not via this plugin. Implement an interface up in your Shared business logic thus:

```
public interface IMyInterface
{
      bool CtrlBluetoothAdapter();
}
```

implement this interface down in your Xamarin.Android code:

```
[assembly: Xamarin.Forms.Dependency(typeof(MyInterface))]
namespace MyApp.Droid.InterfaceImplementations
{
    public class MyInterface: IMyInterface
    {
        public bool CtrlBluetoothAdapter(bool fEnable)
        {
             try
            {
                var bluetoothAdapter = BluetoothAdapter.DefaultAdapter;
                //todo check BluetoothAdapter.IsEnabled first
                ((fEnable)?bluetoothAdapter.Enable():bluetoothAdapter.Disable());
                return true;
            }
            catch (Exception e)
            {
                return false;
            }
     }
}
```

Note, you will need to implement the interface in Xamarin.Ios as well even if it does nothing but return false.

Karen

bnserig, 2019-11-07:
Your answer confirm that is not possible with this plugin.
I didn't use a interface implementation but I use SimpleInjector to do that.
Thanks.

closed 2019-11-07 by bnserig

veletron2, 2019-11-07:
SimpleInjector looks nice. I've got a whole load of methods such as the above, so the interface seemed to be the simplest way to include them all. These are obvious things to end up in Xamarin.Essentials someday, but currently that seems restricted to things for which a multi-platform solution exists and there so much that you just cannot do with iOS.

bnserig, 2019-11-07:
My project is Android only so I don't have iOS limitations.
SimpleInjector is usefull when you have projects without Xamarin.Forms reference.