dotnet-bluetooth-le/dotnet-bluetooth-le #404: Reading data from characteristics is not working

srishalu, 2019-12-20:
I am using BLE device with respect to Xamarin.Forms. Can you please help to resolve the below issue? 

## Steps to reproduce

1. After scanning I have some set of devices . I am selecting the device. After selecting, i am connecting to that device.  

2. I got some list of services. so i am iterating one service  and i got that service characteristics

3. I got some list of characteristics. I am iterating first characteristics and reading. Application crashed.

## Expected behavior
Tell us what should happen

## Actual behavior
Instead of crash, it should read and give bytes

### Crashlog
Excepotion is not thrown. These are the details i got in output screen

12-20 17:54:18.170 D/BluetoothGatt(20372): onConnectionUpdated() - Device=98:07:2D:45:76:83 interval=39 latency=0 timeout=500 status=0
12-20 17:54:22.020 D/BluetoothGatt(20372): onConnectionUpdated() - Device=98:07:2D:45:76:83 interval=798 latency=0 timeout=1000 status=0
12-20 17:54:22.290 I/Choreographer(20372): Skipped 248 frames!  The application may be doing too much work on its main thread.
12-20 17:54:22.292 W/Looper  (20372): Slow Frame: doFrame is 4148ms late
12-20 17:54:24.763 F/        (20372): * Assertion at /Users/builder/jenkins/workspace/xamarin-android-d15-9/xamarin-android/external/mono/mono/mini/debugger-agent.c:4931, condition `is_ok (error)' not met, function:get_this_async_id, Could not execute the method because the containing type is not fully instantiated. assembly:<unknown assembly> type:<unknown type> member:(null)

## Configuration

**Version of the Plugin:** e.g. 2.1.1 (Latest)

**Platform:** Android  8.1 API - 27

**Device:** Redmi Note

**Xamarin.Forms:** 4.4.0.991265(Latest)

I couldnt attach sample. Here are my snippets

```
private async void Button_Clicked(object sender, EventArgs e)
        {
            var services = await _connectedDevice.GetServicesAsync();
            if (services != null)
            {
                var characteristics = await services[0].GetCharacteristicsAsync();
                if (characteristics != null)
                {
                    foreach (var character in characteristics)
                   {
                                var byteValue = await character.ReadAsync(); /// Crashed while reading this line
                                 var str = Encoding.UTF8.GetString(byteValue);
                                 output.Text = str;
                            }                  
                }
            }
}
```

Thanks in advance

alejandropadillav1, 2020-09-09:
Do you have any idea about that?, I have the exact error and the exact line..

Thank you

alejandropadillav1, 2020-09-09:
I've tested with the master branch, and it works with no error for your information, It is so weird.

Thank you.

nikolayDemirev, 2022-04-10:
I have the same issue. Do you have any updates on it?