dotnet-bluetooth-le/dotnet-bluetooth-le #405: Adapter.StartScanningForDevicesAsync() crashed on iOS 13.3

janlange, 2020-01-14:
This simple code makes the iOS Xamarin app crash on iOS 13.3, it worked fine on iOS 12.x

adapter = CrossBluetoothLE.Current.Adapter;

// THIS LINE MAKES THE APP CRASH:
await adapter.StartScanningForDevicesAsync();

foreach (IDevice idev in adapter.DiscoveredDevices)
...

Plugin 2.2.0-pre2
iOS 13.3 
IPhone 10 XS MAX

Tacomatt, 2020-01-15:
I just updated my iPhone XS to 13.3.1 Public Beta 2 and am seeing the same behavior. It did the same thing when I was on Public Beta 1 as well as with the official 13.3 release on an iPhone 8.

janlange, 2020-02-28:
I found the answer in the updated readme:
Add these to info.plist:
<key>NSBluetoothAlwaysUsageDescription</key>
<string>YOUR CUSTOM MESSAGE</string>

closed 2020-02-28 by janlange