dotnet-bluetooth-le/dotnet-bluetooth-le #406: iOS: App crashes upon startup

MKuckert, 2020-01-15:
## Steps to reproduce

Not totally sure how to reproduce this issue and when the OS calls `DisconnectedPeripheral` on the `BleCentralManagerDelegate`. But we observed an `IndexOutOfRangeException` in `ClearServices`. Possibly a race condition because of the lacking lock around `KnownServices`?

## Expected behavior

The app should start normally

## Actual behavior

The app crashes

```
System.IndexOutOfRangeException: Index was outside the bounds of the array.
  at System.Collections.Generic.List`1+Enumerator[T].MoveNext () [0x00023] in <e9fc9f9aa3924b8ab178e9f9b7538a42>:0 
  at Plugin.BLE.Abstractions.DeviceBase.ClearServices () [0x00040] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\DeviceBase.cs:103 
  at Plugin.BLE.iOS.Adapter.<.ctor>b__4_3 (System.Object sender, CoreBluetooth.CBPeripheralErrorEventArgs e) [0x000fb] in <cb56f614a3184500b5c131e49c594b7e>:0 
  at Plugin.BLE.BleCentralManagerDelegate.DisconnectedPeripheral (CoreBluetooth.CBCentralManager central, CoreBluetooth.CBPeripheral peripheral, Foundation.NSError error) [0x00013] in <cb56f614a3184500b5c131e49c594b7e>:0 
  at (wrapper managed-to-native) UIKit.UIApplication.UIApplicationMain(int,string[],intptr,intptr)
  at UIKit.UIApplication.Main (System.String[] args, System.IntPtr principal, System.IntPtr delegate) [0x00005] in /Library/Frameworks/Xamarin.iOS.framework/Versions/13.2.0.42/src/Xamarin.iOS/UIKit/UIApplication.cs:86 
  at UIKit.UIApplication.Main (System.String[] args, System.String principalClassName, System.String delegateClassName) [0x0000e] in /Library/Frameworks/Xamarin.iOS.framework/Versions/13.2.0.42/src/Xamarin.iOS/UIKit/UIApplication.cs:65 
  at APP.iOS.MainApplication.Main (System.String[] args) [0x00001] in /Users/mkuckert/work/projects/APP/src/APP.iOS/MainApplication.cs:9 
```

## Configuration

**Version of the Plugin:** 2.1.1

**Platform:** iOS 13.3

**Device:** iPhone SE and others

MKuckert, 2020-01-17:
I was able to fix this error by adding locking to `DeviceBase.KnownServices`: https://github.com/xabre/xamarin-bluetooth-le/pull/407

logikonline, 2020-06-08:
Can you please post the exact code?

closed 2020-12-29 by MKuckert