dotnet-bluetooth-le/dotnet-bluetooth-le #409: iOS 13.3 Device disconnected while fetching services

thierryvt, 2020-01-22:
The exact same code works fine on Android but on iOS it throws an exception.
Exact code is as simple as:

`ConnectParameters p = new ConnectParameters(false, false);
await adapter.ConnectToDeviceAsync(device, p);
var dataTransferService = await item.Device.GetServiceAsync(Guid.Parse(_bleDataTransferServiceID));`

The connecting works (device state == connected). It fails while trying to get the services.
It makes no difference whether I use GetServiceAsync(<GUID>) or GetServicesAsync, both fail with the same exception.

## Steps to reproduce

1. Connect to a ble tag on ios

2. attempt to get services using var services = await device.GetServicesAsync();
Or even "GetServiceAsync" with a known service GUID

3. observe above exception gets thrown

## Expected behavior
The requested service / services should be returned

## Actual behavior
An exception gets thrown indicating that the device has disconnected while attempting to get the service(s).

### Crashlog

error message: Device disconnected while fetching services

Relevant bit of stacktrace:
at Plugin.BLE.Abstractions.Utils.TaskBuilder.FromEvent[TReturn,TEventHandler,TRejectHandler] (System.Action execute, System.Func`3[T1,T2,TResult] getCompleteHandler, System.Action`1[T] subscribeComplete, System.Action`1[T] unsubscribeComplete, System.Func`2[T,TResult] getRejectHandler, System.Action`1[T] subscribeReject, System.Action`1[T] unsubscribeReject, System.Threading.CancellationToken token) [0x000fd] in <12956cfcb4ff491aa0cc1f7ab64b7e72>:0 
  at Plugin.BLE.Abstractions.DeviceBase.GetServicesAsync (System.Threading.CancellationToken cancellationToken) [0x00050] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\DeviceBase.cs:60 
  at Plugin.BLE.Abstractions.DeviceBase.GetServiceAsync (System.Guid id, System.Threading.CancellationToken cancellationToken) [0x0003f] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\DeviceBase.cs:69 
  at Enrollment.Models.BLE.BleConnectionManager.Connect
...

## Configuration

**Version of the Plugin:** 
2.1.1

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
iOS 13.3

**Device:** 
iPhone 11

Claus-TSI, 2020-05-23:
Has anyone found a workaround for this? I am seeing the same thing in iOS 13.5 on a iPhone 8.

thierryvt, 2020-05-23:
> Has anyone found a workaround for this? I am seeing the same thing in iOS 13.5 on a iPhone 8.

Never found a workaround BUT after some debugging I could conclude it also wasn't an issue with the library. I downloaded the source code so I could debug what was going on and (in our case) it seemed to have been an issue with either the way Apple has implemented the BLE standard or the tag itself.

It was literally a communications error that only ocurs on iOS devices

Claus-TSI, 2020-05-24:
Thanks @thierryvt,
I appreciate you taking the time to respond. 
Cheers,
C

naryk86, 2022-10-17:
Any updates?
It happens something similar in my app... and this behaviour did not happens few months ago [link](https://github.com/xabre/xamarin-bluetooth-le/issues/631)