dotnet-bluetooth-le/dotnet-bluetooth-le #419: StartScanningForDevicesAsync not working

georgemichailou, 2020-03-13:
The StartScanningForDevicesAsync method is not find devices.

https://github.com/georgemichailou/working-with-plugins

## Steps to reproduce

1. Following the documentation (vanilla package)

2. Scan for devices 

## Expected behavior
Find available Devices 

## Actual behavior
Nothing happened. I get every time time out without any result

## Configuration

**Version of the Plugin:** 2.2.0-pre2 and 2.2.1

**Platform:** Android 9 

**Device:** Samsung s8

georgemichailou, 2020-03-14:
The Plugin.BLE.Abstractions.Contracts.ScanMode.LowLatency fixing the problem. Thank you

closed 2020-03-14 by georgemichailou

bbenetskyy, 2020-06-16:
We have just the same problem and ScanMode LowLatency does not fix the problem for us. I will back here when we will find a solution.

naveenchaganti, 2020-07-09:
StartScanningForDevicesAsync is not working even ScanMode LowLatency mode.

Denaindra, 2020-10-24:
@naveenchaganti  have you fixed your issue , me also facing same issue ?