dotnet-bluetooth-le/dotnet-bluetooth-le #420: Problem with scanning

msthrax, 2020-03-24:
https://github.com/msthrax/BluetoothTryApp.git
(I committed the project and it is uploading, if you refer to this git and nothing was here, please just consider those file I uploaded in this note and then check this git later. thanks)

## Steps to reproduce
Hi
First of all, I really thank you your efforts regarding production of this library.
I installed this library through nuget in my Xamarin.Forms project which include Android, iOS and UWP projects. (And also the Core project).
I want to talk about Android project now and iOS and UWP platform is not my goal for now.

1. I have installed the (vanilla) package in my core and also my android project. The version is 2.2.0-pre2.

2. I tried to setup everything myself but it didn't work, so I decided to follow this video that used this library. The link is: https://www.youtube.com/watch?v=IFRqCHLloLA
(which is 4 parts)

3. Targeted Android API is 28 ( Android 9) and minimumAPI is 22 (Android 5.1).

4. I have added needed permissions in my manifest file as below:
`<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" android:versionCode="1" android:versionName="1.0" package="com.companyname.bluetoothtryapp" android:installLocation="preferExternal">
	<uses-sdk android:minSdkVersion="22" android:targetSdkVersion="28" />
	<application android:label="BluetoothTryApp.Android"></application>
	<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
  
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
  
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />
</manifest>`

 5. I have a single page that a simple "Start" button would trigger of every thing(Setting up classes, initialization and Bluetooth Scan).
this is my cs code:
CS File:
[MainPage.xaml.zip](https://github.com/xabre/xamarin-bluetooth-le/files/4375628/MainPage.xaml.zip)
XAML file:
[MainPage.zip](https://github.com/xabre/xamarin-bluetooth-le/files/4375672/MainPage.zip)

## Expected behavior
I expected to see available Bluetooth devices around my self.
I connected a Bluetooth USB Dongle to my Computer and it is active and discover-able.
I have another phone with enabled Bluetooth.
Also i got two gadgets( one is Samsung Level U Pro and another one is a non-OS smart watch).
Level U Pro is paired to my Device to check the "GetPairedDevice" functions.

## Actual behavior
The thing that happened is actually nothing. But the "StateChanged" event and the "State" property were OK.

### Crashlog
Nothing crashed.

## Configuration

**Version of the Plugin: 2.2.0-pre2

**Platform: Android 9.0 - Pie

**Device: Samsung A70 (SM-A705FN/DS)

akadow, 2020-04-15:
Hi All,

I would like to add my thanks for all of the hard work on this project.

Similar to @msthrax, my implementation seems to work (pick up many bluetooth devices around me including a laptop) but no luck on other bluetooth enabled phones or tablets. 

Have tried an iphone5s, and several android devices running Android 7.1 to 9 both running my application and for test.

I wonder if I am missing something or if this is just a limitation of this method?

Thanks again
Alex

msthrax, 2020-04-15:
Hello
I created a working project with my own Android implementation and also Cross platform interface.
You can check it here:
[https://github.com/msthrax/BLEApp]

@akadow

akadow, 2020-04-19:
Awesome, thanks @msthrax, will take a look! Cheers.

akadow, 2020-04-19:
Thats great @msthrax, thanks again. 

I notice this allows any device that is 'discoverable' to be listed. 

I am aware that this can be controlled in Android with the help of BluetoothAdapter.ActionRequestDiscoverable but do you happen to know how to make an ios device discoverable? Have been hunting around for the answer for quite some time.

msthrax, 2020-04-19:
> Thats great @msthrax, thanks again.
>
> I notice this allows any device that is 'discoverable' to be listed.
>
> I am aware that this can be controlled in Android with the help of [...]

You'r welcomed :)

Unfortunately I don't have any experience on iOS. Sorry

bapgobears, 2020-04-25:
I had a similar experience - scan returned no devices.  I discovered I needed to enable Location permissions in the settings for the app.  On the Android device go to Settings/Apps/YourApp and enable the Location permissions.

steeve49, 2020-06-01:
Setting Location on into app settings is working but is that a good solution when publishing your app on a store ?
Is there a way to avoid managing app settings through android parameters and ask for location as the rest of authorizations ?

mikaelsundin, 2020-10-13:
Yes, the program must first check with location service if it enable or not.
Else the scan will not return any result on Android.

This is how i solved it, check if location is enabled. If not tell the user to enable it.
```   
public bool IsLocationEnabled(){
    try
    {
        var loc = (Android.Locations.LocationManager)Xamarin.Essentials.Platform.AppContext.GetSystemService(Android.Content.Context.LocationService);
        return loc.IsLocationEnabled;
    }
    catch (Java.Lang.NoSuchMethodError)
    {
        //If no location service exist.
        return true;
    }
}
```