dotnet-bluetooth-le/dotnet-bluetooth-le #421: Plugin.BLE 2.2.0-pre2 - error in the UWP project when uses  CrossBluetoothLE.Current.Adapter

Auto72, 2020-03-30:
I'm trying to use the Plugin.BLE (version 2.1.1) with a test solution in Visual Studio 2019 and Xamarin Forms. It works in Android and iOS. When I try to run the UWP project, I get the following error when I use this code:

    private readonly IAdapter _adapter;
    public MainPage()
    {
        InitializeComponent();

        _adapter = CrossBluetoothLE.Current.Adapter;    // I get the error here
    }

System.NotImplementedException HResult=0x80004001 Message=This functionality is not implemented in the portable version of this assembly. You should reference the NuGet package from your main application project in order to reference the platform-specific implementation. Source=Plugin.BLE StackTrace: at Plugin.BLE.CrossBluetoothLE.get_Current() at App1.MainPage..ctor() in C:\App\App1\App1\MainPage.xaml.cs:line 28 at App1.App..ctor() in C:\App\App1\App1\App.xaml.cs:line 13 at App1.UWP.MainPage..ctor() in C:\App\App1\App1.UWP\MainPage.xaml.cs:line 24

Then I installed the Plugin.BLE 2.2.0-pre2, but this time I get the following error:

System.IO.FileLoadException HResult=0x80131040 Message=Could not load file or assembly 'Microsoft.Toolkit.Uwp.Connectivity, Version=5.1.0.0, Culture=neutral, PublicKeyToken=null'. The located assembly's manifest definition does not match the assembly reference. (Exception from HRESULT: 0x80131040) Source=Plugin.BLE StackTrace: at Plugin.BLE.BleImplementation.InitializeNative() at Plugin.BLE.Abstractions.BleImplementationBase.Initialize() at Plugin.BLE.CrossBluetoothLE.CreateImplementation() at System.Lazy1.PublicationOnlyViaFactory(LazyHelper initializer) at System.Lazy1.CreateValue() at Plugin.BLE.CrossBluetoothLE.get_Current() at App1.MainPage..ctor() in C:\App\App1\App1\MainPage.xaml.cs:line 28 at App1.App..ctor() in C:\App\App1\App1\App.xaml.cs:line 13 at App1.UWP.MainPage..ctor() in C:\App\App1\App1.UWP\MainPage.xaml.cs:line 24

The Microsoft.NETCore.UniversalWindowsPlatform pakage is the latest version: 6.2.10

Auto72, 2020-04-02:
To investigate more, I used a breakpoint over the line where I get the error in the UWP project:

_adapter = CrossBluetoothLE.Current.Adapter;    // I get the error here

When the execution of the program is there, I selected the "CrossBluetoothLE.Current.Adapter" method and pressed SHIFT-F9 to get the Quick Watch.
The error shown is the following:

error CS0119: 'CrossBluetoothLE' is a type, which is not valid in the given context	

See attached picture.
[image]

I downloaded the source of the plug-in version 2.2.0-pre2 and updated two NuGet packages to the latest version. 

Microsoft.NETCore.UniversalWindowsPlatform
and
Xamarin.Forms

I was able to compile and execute the included sample BLE.Client.
The UWP project of the BLE.Client DOES work fine!
It uses the MVVM version of the Plug-in.

I use the Vanilla version and the following method DOESN'T work:

CrossBluetoothLE.Current.Adapter;

Can you help me, please?
Can the method "CrossBluetoothLE.Current.Adapter" be used with Vanilla plug-in version 2,1,1-pre2?

Thanks.

Auto72, 2020-04-03:
I found the solution to the problem :-)
In the UWP project of your application, you need to reference the following NuGet package:

"Microsoft.Toolkit.Uwp.Connectivity"
Version --> 5.1.0

I hope the people in charge of the Plugin.BLE will release the 2.2.0 version soon.
They also should specify to add add this reference.

Thanks.

naxtester, 2020-11-06:
I am using your Bluetooth LE plugin for Xamarin, but I also have this problem. I can't get the adapter, services and features to write and read.
Could this problem be solved in version 2.2.0?
When do you expect to release the next version 2.2.0? 

My project depends on its use.

Regards.

Auto72, 2020-11-06:
Dear Mr. Xabre,
Could you please fix - at least - just this bug and release a version (Plugin.BLE 2.2.0-pre3) 
in order to allow the Plugin.BLE to be used with UWP, otherwise this world is left out.
Thanks,

phispi, 2022-06-30:
For `Plugin.BLE 2.2.0-pre4` I still needed to add the `Microsoft.Toolkit.Uwp.Connectivity` nuget package (version 6.1.1). When trying to use the latest version 7.1.2 of `Microsoft.Toolkit.Uwp.Connectivity`, I got a `Attempted to access a missing method` exception once calling `StartScanningForDevicesAsync()`. This was also mentioned in https://github.com/xabre/xamarin-bluetooth-le/issues/591.

janusw, 2022-07-27:
> When trying to use the latest version 7.1.2 of `Microsoft.Toolkit.Uwp.Connectivity`, I got a `Attempted to access a missing method` exception

According to https://github.com/xabre/xamarin-bluetooth-le/issues/591#issuecomment-1125306120, this is due to `Windows.System.DispatcherQueue.HasThreadAccess`. As mentioned in https://docs.microsoft.com/en-us/uwp/api/windows.system.dispatcherqueue?view=winrt-22621#version-history, this method was added in SDK version 18362, so I guess one should just update to this SDK version when using Microsoft.Toolkit.Uwp.Connectivity v7 ...

DamianSuess, 2022-09-04:
As per @phispi, downgrading to `Microsoft.Toolkit.Uwp.Connectivity` v6.1.1 worked for me as well.

janusw, 2022-09-06:
> As per @phispi, downgrading to `Microsoft.Toolkit.Uwp.Connectivity` v6.1.1 worked for me as well.

Alright. There are probably two things we should do here:
* in the short term: update the dependency in the nuspec and disallow Uwp.Connectivity v7
* in the longer term: move to Uwp.Connectivity v7

> As an FYI, one thing to watch out for on UWP is the use of Xamarin.Essential's `await Permissions.CheckStatusAsync<Permissions.LocationWhenInUse>();`. It appears that UWP doesn't think it has the permissions when in fact it did. Even requesting the permission still threw, `Denied`.

Is it this problem? https://github.com/xamarin/Essentials/issues/1771 Or something different?

Unfortunately the corresponding PR is open since quite a while: https://github.com/xamarin/Essentials/pull/1777

janusw, 2022-12-04:
> Alright. There are probably two things we should do here:
>
>     * in the short term: update the dependency in the nuspec and disallow Uwp.Connectivity v7
>
>     * in the longer term: move to [...]

Actually the latter has been done for version [3.0.0-beta.2](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/releases/tag/3.0.0-beta.2), see #640.

janusw, 2022-12-04:
AFAICS the original problem in this issue should also be fixed with the 3.0.0 beta versions, at least I cannot reproduce it any more.

@Auto72 If you still have problems with UWP with the latest release, please file a new issue.

closed 2022-12-04 by janusw