dotnet-bluetooth-le/dotnet-bluetooth-le #423: UWP - _adapter.DisconnectDeviceAsync() doesn't work - plugin ver. 2.2.0-pre2

Auto72, 2020-04-03:
## Steps to reproduce

1. UWP application project with BLE Plugin version: 2.2.0-pre2

2. Open a connection with a device 

await _adapter.ConnectToDeviceAsync(_device);
_myService = await _device.GetServiceAsync(wantedServicetUUID);
_myCharacteristic = await __myService GetCharacteristicAsync(wantedCharacteristicUUID));

_myCharacteristic.ValueUpdated += (o, args) =>
{
    var bytes = args.Characteristic.Value;
};
await _myCharacteristic .StartUpdatesAsync();

3. Close the Connection

await _myCharacteristic.StopUpdatesAsync();
await _adapter.DisconnectDeviceAsync(_device);

## Expected behavior
The connection with the bluetooth LE device is closed.

## Actual behavior
The connection with the bluetooth LE device is NOT closed.
The following command, never ends:

await _adapter.DisconnectDeviceAsync(_device);

### Crashlog

No crash occurs.

## Configuration

Version of the Plugin: 2.2.0-pre2

Platform: Windows 10 1909

Device: Windows 10 1909

JongChern, 2021-07-19:
Hi was wondering if there are any immediate plans to get this fix released on nuget

sisaacks, 2021-09-07:
Hello has this been added to Pre2?  I am still seeing this issue

TStein1984, 2021-12-12:
@JanNepras @xabre @janusw @sisaacks if have the same issue. Is there some hack to fix it or another solution so solve the problem for UWP? Thanks in advance.

janusw, 2022-12-10:
Does this still occur with 3.0.0-beta.2?

Auto72, 2023-01-26:
Still doesn't work with 3.0.0-beta.2.

IDevice device;
ICharacteristic fifoCharacteristic;

The following command now crashes in UWP, even if it is surrounded by a try catch block:

await adapter.DisconnectDeviceAsync(device);

Sometimes also the following command crashes on UWP  (called before the previous one):

await fifoCharacteristic.StopUpdatesAsync();

janusw, 2023-08-05:
Fixed via #721 in [3.0.0-beta.5](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/releases/tag/3.0.0-beta.5).

closed 2023-08-05 by janusw