dotnet-bluetooth-le/dotnet-bluetooth-le #429: [Suggestion] Prism resolvable object

Jesus805, 2020-04-29:
Greetings,

I have a Xamarin forms app that uses Prism Library as my dp container. It would be nice if BleImplementation was exposed so that we can register the type without needing to use MVVM cross.

smstuebe, 2020-04-29:
Hi,

you don't need to install the MvvmCross package.
install the vanilla package (Install-Package Plugin.BLE) and then you can get it via `CrossBluetoothLE.Curent`

Jesus805, 2020-04-29:
Appreciate the reply. 

That's the method I'm currently using, which works. However, using CrossBluetoothLE.Current resolves the IBleManager instance during container registration.

Jesus805, 2020-07-27:
Hi,

I'm going to go ahead and close this because I recently learned that you can register a function, so you don't really need to register the type.
In Prism you register it like so:
```containerRegistry.RegisterSingleton<IBluetoothLE>(() => CrossBluetoothLE.Current);```

Thanks

closed 2020-07-27 by Jesus805

DamianSuess, 2023-01-02:
FYI for **Prism.Maui**. If I could make a PR for a wiki page, this would be part of it.

## Code Snippet

```cs
        services.AddSingleton<IBluetoothLE>(CrossBluetoothLE.Current);
        services.AddSingleton<IAdapter>(CrossBluetoothLE.Current.Adapter);
```

## Full Builder Sample (MauiProgram.cs)

```cs
 public static MauiApp CreateMauiApp()
  {
    var builder = MauiApp.CreateBuilder();
    builder
      .UseMauiApp<App>()
      .UsePrism(ConfigureContainer)
      .ConfigureFonts(fonts =>
      {
        fonts.AddFont("OpenSans-Regular.ttf", "OpenSansRegular");
        fonts.AddFont("OpenSans-Semibold.ttf", "OpenSansSemibold");
      });

    return builder.Build();
  }

  private static void ConfigureContainer(PrismAppBuilder builder)
  {
    builder
      .ConfigureServices(services =>
      {
        services.AddSingleton<IBluetoothLE>(CrossBluetoothLE.Current);
        services.AddSingleton<IAdapter>(CrossBluetoothLE.Current.Adapter);
      })
      .RegisterTypes(container =>
      {
        container.RegisterForNavigation<MainView, MainViewModel>();
        container.RegisterInstance(SemanticScreenReader.Default);
      })
      .OnAppStart(async navService =>
      {
        var result = await navService.NavigateAsync($"{nameof(NavigationPage)}/{nameof(MainView)}");
        if (!result.Success)
        {
          System.Diagnostics.Debugger.Break();
        }
      });
  }

```