dotnet-bluetooth-le/dotnet-bluetooth-le #432: [Question] Possible For Android Phone & iPhone to See Each Other?

stefanhk31, 2020-05-05:
## Steps to reproduce

1. Install Sample app on an Android phone and and iPhone

2. Put phones next to each other and make sure they both have Bluetooth turned on

3. Run app on Android phone and scan

4. Run app on iPhone and scan
 
## Expected behavior
On Android phone, iPhone should be visible in list of nearby devices
On iOS, Android should be visible in list of nearby devices

## Actual behavior
Android cannot see iPhone, and iPhone cannot see Android phone. This may be a limitation from Google and/or Apple on what data devices broadcast and to whom, or it may be something that can be addressed in the plugin. But I thought that it was at least possible to view, if not connect to, devices across Android/iOS. 

### Crashlog

N/A

## Configuration

**Version of the Plugin:**  2.1.1
**Platform:** iOS 13.4.1, Android 10.0 API 29

**Device:** iPhone SE, Google Pixel 3

StephanRaab, 2020-07-09:
@stefanhk31 That would not be the expected behavior. This is `BLE` and I believe you're thinking of `Classic Bluetooth`.

ymihir, 2020-11-12:
Is it possible to read data without pairing each of them? like reading other phone's same app data like appid, or some unique keys that app is having?