dotnet-bluetooth-le/dotnet-bluetooth-le #433: Unable to Read Characteristics on iOS 13

bensuarezbrodie, 2020-05-08:
On any device running iOS 13. 
I connect to the device and am able to read the list of services from it.
However, as I go to either GetCharacteristicsAsync or GetCharacteristicAsync(GUID)
I get disconnected with an error message of
Device UDID "disconnected while fetching characteristics for service" GUID

This does not happen on iOS 12 or under devices

The snippet of code

// Retrieving services
            while (!gotServices)
                try
                {
                    listOfServices = await device.GetServicesAsync();
                    gotServices = true;
                }
                catch(Exception e)
                {
                    Console.WriteLine(e.Message);
                }

// Loop through each service
foreach(var service in listOfServices){

// Retrieve Characteristic
while (!gotCharacteristics)
            {
                try
                {
                    var characteristic = await service.GetCharacteristicsAsync();
                    gotCharacteristics= true;
                }
                catch (Exception e)
                {
                    Console.WriteLine(e.Message); // Throws error here
                }
            }
}
## Steps to reproduce

1. Connect to Device

2. Read a list of services

3. Try to read Characteristic from any of the services.

## Expected behavior
For it to return an ICharacteristic Object

## Actual behavior
A Catch with an exception message of 
Device UDID "disconnected while fetching characteristics for service" GUID

### Crashlog

None

## Configuration

**Version of the Plugin:** 2.1.1 / 2.0.0 / 2.2.0-pre2

**Platform:** iOS 13.3.1

**Device:**  iPhone 8

majdaneq, 2020-05-21:
Hi @bensuarezbrodie have you found any workaround? I'm facing the same issue.

bensuarezbrodie, 2020-05-21:
@majdaneq 

Yes actually. It was an issue with iOS 13.3.1

13.4 was released here a week or so ago, and I was able to fetch characteristics after that.

Check that you can upgrade to the latest version?

closed 2020-05-21 by bensuarezbrodie

majdaneq, 2020-05-21:
@bensuarezbrodie 

unfortunately still facing this issue with 13.4.1