dotnet-bluetooth-le/dotnet-bluetooth-le #435: Reading characteristic returns wrong first 22 characters

tagtraeumer, 2020-05-19:
I'm trying to receive a json configuration from my BLE device and use it in my app. but very often the first attempt of receiving the configuration the first 22 characters contain information of another json string that was used previously. I checked the BLE message with nRF Connect and the device always sends the correct string (which is also logged on the device itself). Debugging my Xamarin App i can also see that the bytes received correlate with the faulty configuration, so in my opinion the chances are high that the problem might be in the plugin itself.

you can find a more detailed error description here:
https://stackoverflow.com/questions/61566065/xamarin-ble-messes-up-received-bytes

## Steps to reproduce

after retrieving the necessary characteristic i call 
`var bytes = await characteristic.ReadAsync();`

## Expected behavior
bytes representing the correct and expected json configuration sent from my BLE Device:
> {"MODE":0,"DEVICE_ID":123456789,"DEVICE_NAME":"Test Name", ... etc

## Actual behavior
The first 22 characters of the json are wrong and might be a residue of another json that i previously sent to the BLE device
> {"DEVICE_NAME":"Test N123456789,"DEVICE_NAME":"Test Name", ... etc

## Configuration

**Version of the Plugin:** 2.1.2

**Platform:** Android 10.0

**Device:** OnePlus 6

boggybrn, 2021-04-10:
I'm seeing something similar...
If I reset my bluetooth device then the connection is broken and my code is informed via the DeviceConnectionLost. Then after scanning and re-establishing a new connection all works well, except....

Every now and then in place of the expected received message I see the last received message from the connection that was previously lost. I have been searching high and low in my code for where this old message might be coming from but as far as I can tell it comes from the plugin.

My code to receive data is:

                        receiveCharacteristic.ValueUpdated += (o, args) =>
                        {
                            var receivedBytes = args.Characteristic.Value;

                            foreach (byte c in receivedBytes)
                            {
                                // do stuff here to buffer & find messages within the data
                            }
                        };

What I see is that receivedBytes sometimes holds the data from the last message received before the connection was lost then re-established.

Version of the Plugin: 2.1.1

Platform: both iOS and Android

janusw, 2021-04-10:
> ## Configuration
> 
> **Version of the Plugin:** 2.1.2
> 
> **Platform:** Android 10.0
> 
> **Device:** OnePlus 6

@tagtraeumer There is no versin 2.1.2 yet. You probably mean 2.1.1?

boggybrn, 2021-04-10:
I tracked down the issue I was having...  When the connection was lost I was not unsubscribing to the ValueUpdated.  So when the new connection was made this also registered a handler.

Then when data arrived from the bluetooth device the handler was called twice - once from the original receiveCharacteristic object (yielding the old message) and once from the receiveCharacteristic object created after the reconnection. After making sure that the handler is unsubscribed before a new receiveCharacteristic object is created all works as expected.

So this looks like something to be wary of rather than a bug in plugin.BLE! Sorry for any intrusion here, but I hope that this may help someone at some point.

AlexanderSCK, 2021-04-12:
Hello, would you please advice me how you did the unsubscribing part and where in the code. Thank you

boggybrn, 2021-04-13:
I moved the code to run on ValueUpdated into a named method:

 private void HandleIncomingCharacters(object o, CharacteristicUpdatedEventArgs args)
 {
        var receivedBytes = args.Characteristic.Value;
        foreach (byte c in receivedBytes)
        {
                    // code to do something useful here
        }
}

The the subscription becomes:
receiveCharacteristic.ValueUpdated += HandleIncomingCharacters;

and unsubscribing becomes:
receiveCharacteristic.ValueUpdated -= HandleIncomingCharacters;

To handle the case where the bluetooth connection is lost because of the remote end I registered a method to receive DeviceConnectionLost events:
 
        bluetoothAdapter.DeviceConnectionLost += OnDeviceConnectionLost;

This method then does the unsubscribing - 

private void OnDeviceConnectionLost(object sender, DeviceEventArgs e)
{
        if(receiveCharacteristic != null)
                receiveCharacteristic.ValueUpdated -= HandleIncomingCharacters;
}

I also unsubscribe if the connection is closed by actions of the user of the app.

AlexanderSCK, 2021-04-19:
Hey, thank you for the answer. So you are not using the StopUpdatesAsync function,  I was wondering if I need to use it or to do it your way. I am sorry I am new to Xamarin and I am wondering when I should do the unsubscribing, because I have multiple devices connected, that need to receive updates. I put the subscribing handler nad StartUpdatesAsync in a method that accepts one parameter of type IDevice and through a loop of already exsiting List of IDevices I call the method multiple times. Can you suggest me a proper approach. Thank you in advance!

`  private async void CollectData(IDevice device)
        {
           
            await _bluetoothAdapter1.ConnectToDeviceAsync(device);
           

            try
            {
                var getDataService = await device.GetServiceAsync(Guid.Parse(serviceID));
                if (getDataService != null)
                {
                    var getDataCharacteristic = await getDataService.GetCharacteristicAsync(Guid.Parse(charID));

                    getDataCharacteristic.ValueUpdated += (o, args) =>
                    {
                        var bytes = args.Characteristic.Value;
                     
                        gatheredInfo.Add(bytes);
                    };

                    await getDataCharacteristic.StartUpdatesAsync();
                }
            }
            catch
            {
            }
        }`

krcenov, 2024-10-30:
Hey, did you guys resolve this?