dotnet-bluetooth-le/dotnet-bluetooth-le #437: Is this plugin supports for Visual studio 2019 Xamarin Projects

ShaileshBhatNP, 2020-06-05:
Is this plugin supports for Visual studio 2019 Xamarin Projects?

Hi, I need to build an Xamarin app using VS2019 and this plugin. Is it possible?. I was confused, after searching for a day..If yes, Can I get the sample in VS2019..I have tried with many samples, but no one is supporting/running in VS2019

ShaileshBhatNP, 2020-06-05:
Where is the template..can you please send me the link..?

ShaileshBhatNP, 2020-06-19:
https://github.com/didourebai/BLEPluginDemo

Got above template which used this plugin..But Failing to Connect

xabre, 2020-06-19:
Please check out the soruce code and run the Sample project. Everything should be fine with VS2019. 

For your own project follow the insturctions in the readme and install the nuget package to all platform specific csprojects and also you shared core csproj.

ShaileshBhatNP, 2020-06-23:
Ok. Thank you, we have to open Source folder for getting sample project(after cloning outer folder)..

closed 2020-06-23 by ShaileshBhatNP