dotnet-bluetooth-le/dotnet-bluetooth-le #440: How to send a Command/Text/Image, after Scanning

ShaileshBhatNP, 2020-06-23:
I'm able to scan the nearby devices using the sample. While clicking connect, its showing connected. But didn't received any request in bluetooth device. Also, How to send a Command/Text/Image to nearby BLE/Classic Bluetooth device, after Scanning..

https://github.com/msthrax/BLEApp 

https://github.com/didourebai/BLEPluginDemo

Above samples also not meet the requirement..Posted issues there also..

StephanRaab, 2020-07-08:
@ShaileshBhatNP 
Could you explain what you're thinking the expected behavior should be?

Remember that ble and classic aren't the same thing.

ShaileshBhatNP, 2020-07-09:
If I need to connect to Mobile from my mobile(mobile phone-mobile phone), is it possible?

If not, what else devices I can connect and send-receive data?..

StephanRaab, 2020-07-09:
This library is meant to be used to pair to `Bluetooth Low Energy` devices.

A mobile phone would be a `Classic Bluetooth Device`. I think what you'll need to do is look into Android/iOS Bluetooth Pairing. Searching for that is what will get you the answer you're looking for.