dotnet-bluetooth-le/dotnet-bluetooth-le #445: iOS Adapter does not properly parse Advertisements for Service UUIDs

jaredballen, 2020-07-10:
There is an issue in the iOS Adapter implementation of Xabre where the CBAdvertisement is not properly parsed for Service UUIDs. iOS lumps all the Service UUIDs received into a single CBAdvertisement regardless of whether or not they are 128-bit, 32-bit, or 16-bit. The UUIDs are placed in an NSArray of NSData (essentially making a jagged array). Xabre is putting all the data into a single list a then flattening it and only emits the advertisements to the upper layers as AdvertisementRecordType.UuidsComplete128Bit. So this means that the data in the advertisement records received, while using Xabre to perform the scan, could be any combination of 128-bit, 32-bit, and 16-bit service UUIDs that a device is advertising for.

## Steps to reproduce

1. Use nRFConnect to create a new Advertiser on a device. Name the device: "Test UUIDs". Set the device up to advertise its name and two 16 bit services UUIDs (1234, 2345).

2. Use the Xabre BLE.Client App on an iOS device

3. Set a break in the OnDeviceDiscovered method of the DeviceListViewModel.cs

4. Continue through each device discovery until you find the device that you set up for advertising in nrfConnect.

5. The Service UUIDs will be in a single advertisement record with 4 bytes. of data. IT should be two records, each with 2 bytes of data and a type of UuidsComple16Bitor UuidsIncomple16Bit.

## Expected behavior
Xabre needs to individually pares each NSData object in the CBAdvertisement and check its length. Then the appropriate advertisement type should be created (128, 32, 16) and forwarded to the upper App layers. That way each service UUID is received individually.

## Actual behavior
The advertisement that is received from Xabre is always a UuidsComplete128Bit with the Data being the size the total bytes from all service UUIDs actually advertised by the device.

## Configuration

**Version of the Plugin:** 2.1.1

**Platform:** iOS 13.3.1

**Device:** iPhone 8 Plus

closed 2020-07-14 by xabre

vecalion, 2021-04-10:
@xabre, I wonder if this fix was released? I'm on v2.2.0-pre2 and here is what I see:

Android as a reference:
[image]

iOS: 
[image]

Notice that all advertised services on iOS are under UuidsComplete128Bit which makes it impossible (?) to parse.

janusw, 2021-04-10:
> @xabre, I wonder if this fix was released? I'm on v2.2.0-pre2 and here is what I see:

The fix has not been released yet. Neither 2.1.1 nor 2.2.0-pre2 include it. But we're planning to prepare a release 2.1.2 soon which will include this ...

vecalion, 2021-04-10:
@janusw thanks for letting me know.

vecalion, 2021-04-11:
@janusw, I wonder if you have some kind of nightly builds? So I can test my code against until the 2.1.2 is released.

janusw, 2021-04-11:
> @janusw, I wonder if you have some kind of nightly builds?

I don't. There is an automatic build set up on Bitrise, but I'm not aware that it produces a nupkg (the build is not public and I don't have full access to it right now). Maybe @xabre can comment on that?

What I could possibly do is to set up an automatic build via GitHub Actions that spits out a nupkg for every new commit on master. Should be pretty straightforward (recently did that for another project).

Also, it's rather easy to build a nupkg yourself, as documented [here](https://github.com/xabre/xamarin-bluetooth-le#how-to-build-the-nuget-package). Those instructions were broken for quite a while, but after my fixes from #493, they should be working again.

vecalion, 2021-04-11:
I did, that was indeed really easy. But I'm getting a mysterious build error when trying to use the nuget that I've created (or .dll, or BLE projects directly) in my project.

```/Users/valeriykovalenko/.nuget/packages/xamarin.forms/5.0.0.2012/buildTransitive/Xamarin.Forms.targets(3,3): Error MSB4018: The "XamlCTask" task failed unexpectedly.
Mono.Cecil.Cil.SymbolsNotMatchingException: Symbols were found but are not matching the assembly
```

Locally and on a clean CI machine. Everything compiles fine when I reference the official xamarin-bluetooth-le package. So I was hoping to check on a nightly build to see if this is something on my side or not.

Sorry for going off topic.

jaredballen, 2021-04-11:
Sorry to continue off topic but a nighty build would be awesome. My company
had to fork the repo into BitBucket to get this change and we're adding
first class support for bonding/pairing on Android as well as automatic
gatt-cache clearing. We'd love to contribute back but everything we pull in
has to be a nuget.

janusw, 2021-04-11:
> I did, that was indeed really easy. But I'm getting a mysterious build error when trying to use the nuget that I've created (or .dll, or BLE projects directly) in my project.
>
> ``` [...]

I actually played that game myself recently and had no problems.

But I also know that you can get into trouble with caching etc when using custom-built local packages. You should at least modify the package version, I think.

> Sorry for going off topic.

No worries! You made a valid point.

janusw, 2021-04-11:
> Sorry to continue off topic but a nighty build would be awesome.

I agree that automatically-built packages would be nice. I'll discuss the status of the Bitrise builds with Adrian, and might possibly look into Github Actions for that purpose.

> My company had to fork the repo into BitBucket to get this change and we're adding first class support for bonding/pairing on Android as well as automatic gatt-cache clearing. We'd love to contribute back but everything we pull in has to be a nuget.

That would be nice. This repo is suffering from too many forks recently.

janusw, 2021-04-11:
> > Sorry to continue off topic but a nighty build would be awesome.
> 
> I agree that automatically-built packages would be nice. I'll discuss the status of the Bitrise builds with Adrian, and might possibly look into Github Actions for that purpose.

Also I have opened #501 for this, in order to avoid going too far off-topic in this dead issue.

janusw, 2021-04-12:
> > > Sorry to continue off topic but a nighty build would be awesome.
> > 
> > 
> > I agree that automatically-built packages would be nice. [..]
> 
> Also I have opened #501 for this

This ticket is solved by now. The nuget packages for the current master branch are available here:
https://github.com/xabre/xamarin-bluetooth-le/suites/2478813162/artifacts/53361185

@vecalion @jaredballen Could you please try those packages and let me know whether they work for you?

PR #446 (2020-07-14): Added iOS/Mac support for 32-bit and 16-bit Service UUIDs
Fixes #445
commit 568f3407 (2020-07-14): Added support for 32-bit and 16-bit Service UUIDs (#446)

Fixes #445