dotnet-bluetooth-le/dotnet-bluetooth-le #447: Reconnect ble device when disconnected

PrachiKarapurkar, 2020-07-20:
How to automatically reconnect ble device when disconnected ?

Abesoddy, 2020-08-05:
Hello, i'm also interested to know a solution to reconnect automatically a ble device with this plugin.

Thanks !!

dariofresu, 2020-11-10:
Have you guys solved the problem somehow?
@PrachiKarapurkar @Abesoddy

PrachiKarapurkar, 2020-11-10:
> Have you guys solved the problem somehow?
> @PrachiKarapurkar @Abesoddy

No, haven't got any solution yet.

dariofresu, 2020-11-12:
> > Have you guys solved the problem somehow?
> > @PrachiKarapurkar @Abesoddy
> 
> No, haven't got any solution yet.

Reading the forums I found a turn around that might help you depending on your application, but I wouldn't recommend it for production purposes.

I do the following:

1.  When creating the model subscribe to the following:   

            //Notify when there is no connection
            bleAdapter.DeviceDisconnected += OnDeviceDisconnected;
            //Notify when connection is lost
            bleAdapter.DeviceConnectionLost += OnDeviceConnectionLost;

2. During the first connection save your device in a variable:

               //connect the device
               await bleAdapter.ConnectToDeviceAsync(model.DispositiveConnected, new ConnectParameters(true), source.Token);
               BLE_device = model.DispositiveConnected;

3. For when you lose the connection and OnDeviceConnectionLost gets called then proceed as follows:

- Check if the device is disconnected
- If disconnected, clear the list of devices
- Start scanning
- Stop scanning as soon as you find the BLE_device saved previously
- Reconnect to the device
- Clear all the services 
- Look for new service 
- Clear characteristics
- Look for the new characteristics
- Subscribe to characteristic notification

Basically, you do it has if the first time you connect it.

Hope it helps,
Dario