dotnet-bluetooth-le/dotnet-bluetooth-le #450: Android - ConnectToDeviceAsync never completes, connecting to a different device 'fixes' issue

IainS1986, 2020-08-17:
I've ran into a periodic issue I see from time to time. I'm doing lots of builds, DEBUG and RELEASE and generally have no issues. I have a couple of devices I connect too.

Sometimes, I find on Android that I can no longer connect to a given device. I can scan fine, see the device, but any call to ConnectToDeviceAsync or ConnectToKnownDeviceAsync just never completes....no errors are thrown and the CancellationToken timeout eventually just fires.

If I try to connect to another device (not the previous one used), that works...then trying to reconnect to the previous device that never completed - that then works and everything is 'fine' again.

Has anyone seen anything similar?

Heres the trace output when trying to connect to a device that never completes (DeviceConnected never fires)...eventually the cancellation token timeout fires.

`[Info] (BluetoothConnectionService.ConnectAsync) Lock Waiting
[Info] (BluetoothConnectionService.ConnectAsync) Lock Acquired
[BluetoothManager] getConnectionState()
[BluetoothManager] getConnectedDevices
[BluetoothGatt] connect() - device: CE:9F:97:95:BE:46, auto: true
[BluetoothGatt] registerApp()
[BluetoothGatt] registerApp() - UUID=5402ae97-da5d-4721-9f97-6ca17c077b43
[BluetoothGatt] onClientRegistered() - status=0 clientIf=11
[OpenGLRenderer] endAllActiveAnimators on 0x70245ca100 (RippleDrawable) with handle 0x70225e7900

<-- Cancellation Token duration expires -->

[BluetoothGatt] cancelOpen() - device: CE:9F:97:95:BE:46
Failed to evaluate expression in conditional breakpoint. Unknown identifier: inflaterFailed to evaluate expression in conditional breakpoint. Unknown identifier: inflater[Info] (BluetoothConnectionService.ConnectAsync) Lock Releasing
[Info] (BluetoothConnectionService.ConnectAsync) Lock Released
`

IainS1986, 2020-08-24:
No solution as of yet. Are you experiencing the same issue?

Don't know if this repository is no longer supported anymore.

IainS1986, 2020-08-24:
Had the issue occur again, the only consistent thing I could see between both instances of the bug occurring was that I didn't connect to the device (or use the app) for around a week (holiday).

I've forked the repo, pulled in the code and can see in Device.cs this is executed as expected...

`var connectGatt = BluetoothDevice.ConnectGatt(Application.Context, autoconnect, _gattCallback, BluetoothTransports.Le);`

However the GattCallback.cs `OnConnectionStateChange` is never fired.

This continued even with deleting and reinstalling the app.

I tried enabled Bluetooth HCI Snoop log in the developer options, and frustratingly that fixed the issue for now! (so the log is likely not going to help as it worked). I'll keep an eye out but I think it will happen again if I leave the app for "some time", which doesn't happen during normal development hence why I've not really seen this bug occur.

YBENZAKOUR, 2020-08-26:
@IainS1986 Hello, Sorry for the later answer, So I have search about this. I have found why this problem. That's because bluetooth le (bluetooth low energy) can't connect with another devices if the last one use bluetooth classic. try to connect your app with bluetooth casque or microphone or another device use ble and it's working .
if you want connect it with another phone for exemple you need to use bluetooth classic ... For android he work without any problem...but for IOS i can't found any solution to connect it 
Sorry because I'm not good in English

IainS1986, 2020-08-27:
I don't think thats the issue i've been facing.

I have a device thats built in house, call it Device A. I have another identical device built in house, call it Device B.

The app connects fine to Device A all* of the time, Debug builds, Release builds, etc.

I leave the app for a week (not confirmed this is definitely the way to reproduce it).

I boot the app up again, and try to connect to Device A (the last device it connected too and *no other connection to Device A has occurred since*. It fails.

The app no longer connects to Device A. The ConnectToDeviceAsync just never completes and the Gatt callback never fires. 

I try to connect to Device B, works first time.

I then try to connect to Device A again, works first time, everything is working as normal again.

IainS1986, 2020-08-27:
Update:

Ok, further investigation after what you just said. Its not time, you're right that it's maybe connecting to another device breaks it. I thought it was leaving it for *some time*, but during that time I connected my phone to some headphones and a car bluetooth dongle. *That was what broke it*.

Can reproduce the issue.

- App connects to Device A fine.
- Disconnect app from Device A and close app.
- I connect my phone to some Headphones (not sure what Bluetooth layer it uses).
- Disconnect headphones.
- Boot up app.
- Try to connect to Device A as before.
- App no longer connects to Device A.
- Connect to Device B, works fine.
- Connect to Device A, works fine.

YBENZAKOUR, 2020-08-27:
OMG, it's verry weird ... for me , I have ESP32 (It's my device A ), she connect without any problem in IOS . she write and read fine... but for my Android phone, He don't found My ESP32 with this plugin!! I don't know why!!! and when I try to connect it with another app like you, He work fine without problem in IOS and Android, Now i try to use another Nuget it's https://github.com/shinyorg/shiny 
So for you, if you want connect just with android, I can help you how do it with bluetooth serial( CLASSIC), But with IOS me too i'm working for solution. 
If you have any news about connexion tell me... And me too I do.

YBENZAKOUR, 2020-08-27:
@IainS1986 I used https://github.com/shinyorg/shiny she work fine without any problem for me in ios and android... try to use it. If you have any problem Tell me . 
Good Luck

IainS1986, 2020-09-04:
I think we have different issues @YBENZAKOUR as you were saying you could *never* connect with Android, whereas mine is all working fine 99% of the time but I was (somehow) getting into a state where the app would no longer connect to a single device.

For others (if anyone reads this), I'm experimenting with a solution that I can't confirm works until I maybe (or maybe don't) see this issue reproduce.

I've turned off AutoConnect and ForceBLE in the connection parameters. Firstly, I believe our device isn't dual mode so ForceBLE is maybe fine, but I'm wondering if AutoConnect was breaking somehow internally.

I've set it to false, BUT FYI this does trigger more Gatt 133 issues sporadically so I've had to implement a simple retry system into our bluetooth task handling.

To combat Gatt 133...

* Ensure you're making all bluetooth calls on the Main Thread
* Ensure you are doing them synchronously. Do NOT make 2 Bluetooth calls at the same time, or overlap them in any way. Block any Bluetooth calls from starting until the current one is finished.
* Do NOT start connecting while you are still scanning
* You will STILL get them occasionally, so do a retry system on connecting. So far I've never had it retry more than 3 times to connect.

I'll see if I get the old issue again, but it could be something deep within the OS bluetooth handling on my phone (Moto G6+) with the AutoConnect parameter :shrug:

IainS1986, 2020-09-10:
So, an update after another week has passed.

My current app works as such,

Scan -> Connect.  then after doing that once with a device and getting the GUID, it just goes straight to Reconnect.

Force BLE is OFF and AutoConnect is OFF. (Unlike before).

A week passed, booted the app up, a Reconnect call was started and....no luck. It wouldn't connect.

But going to do another scan, then connect DID work (which before would also *not* work until I connected to a new device).

So google around a bit more I think I might have figured out whats going on, and think it comes down to just Androids inconsistencies with BLE across manufacturers.

These Medium articles helped shed some light, https://medium.com/@martijn.van.welie/making-android-ble-work-part-1-a736dcd53b02 and https://medium.com/@martijn.van.welie/making-android-ble-work-part-2-47a3cdaade07

Android has an internal BLE Cache which can be cleared *at some point*. Sometimes when Bluetooth is turned off, sometimes when the phone is turned off, or in my phones case (Moto G6+)....just 'some other time'!

When this happens, using the AutoConnect parameter **will not work**. Android, at least on Moto, cannot autoconnect to a previously known device if its no longer in the OS Cache. Even though the GUID is right, the internal cache is empty.

Similarly, when using Reconnect API, it builds a BluetoothDevice from the GUID but it can't connect to this if the OS Cache is empty! The medium article mentions you *can* check for this issue by looking at the BleutoothDevice deviceType to see if its 'Unknown', in which case you can't reconnect. Otherwise a new Scan + Connect should work (and does in my instance).

So. I think this is the issue. I'm going to do another test.

I'm going to switch ForceBLE back on as I think this will mitigate some of the Gatt 133 errors I'm seeing as they may be occuring when it tries to use Classic. 

I'm then going to remove calls to Reconnect and instead do a Scan -> Connect loop every time....this is fine for my use case to be fair.

I'll have 2 apps installed on my phone, one that tries to Reconnect, one that does a fresh Scan -> Connect. 

We'll see in another week (or if I can figure out how to clear my phones BLE cache properly) to see if App A can't reconnect but App B can and if so, I think thats the culprit!

IainS1986, 2020-09-17:
Update:

Another week passed. Checked the two apps.

App A: AutoConnect=FALSE, called ConnectToKnownDevice -> Failed to connect
App B: AutoConnect=FALSE, did a scan and then ConnectToDevice -> Connected

So far it seems like my suspicions around trying to connect to a known device when the internal OS BLE cache has been cleared does not are correct.

I also suspect this repo is potentially abandoned at this point? Either way, anyone who runs into this...

**If you use AutoConnect=TRUE, if the device BLE cache is cleared, ConnectToDevice and ConnectToKnownDevice can fail to ever connect**

**If you use ConnectToKnownDevice, this can fail to connect if the device BLE cache is cleared, regardless of AutoConnect value**

**My advice, avoid both if you're not sure, otherwise you will have to do your own handling for fallback to Scan->Connect. I might look to add functionality to this plugin to have a 'check can reconnect' function in Droid that checks if the Device object type is Unknown or not, as that (apparently) lets you know if you can do a ConnectToKnownDevice call or not**

I might report back periodically to confirm my findings, or if they change.

IainS1986, 2020-10-05:
Update for anyone - more time has passed and i've seen the versionof the App that tries to do a ConnectToKnownDevice call fail multiple time while the version of the app that does a fresh Scan + ConnectToDevice call work fine.

Pretty confident I've confirmed it to be Android internal BLE cache clearing out, and there for ConnectToKnownDevice will *fail* and AutoConnect=TRUE will *fail*.

Propose maybe having a 'CanConnectToKnownDevice' call that for iOS etc just returns TRUE and for Android does a check on the device type to see if its Unknown (not cached), similar too this Android code...

`// Get device object for a mac address
BluetoothDevice device = bluetoothAdapter.getRemoteDevice(peripheralAddress)
// Check if the peripheral is cached or not
int deviceType = device.getType();
if(deviceType == BluetoothDevice.DEVICE_TYPE_UNKNOWN) {
    // The peripheral is not cached
} else {
    // The peripheral is cached
}`