dotnet-bluetooth-le/dotnet-bluetooth-le #454: Scanning doesn't see any devices

rafaelagp, 2020-08-31:
Scanning works fine when compiling with SDK 28 but fails to find devices with any SDK upwards when running on Android 10.

## Steps to reproduce

1. Call StartScanningForDevicesAsync()

2. DeviceDiscovered is never called

3.

## Expected behavior
Callback DeviceDiscovered is run whenever a device is found after calling StartScanningForDevicesAsync().

## Actual behavior
StartScanningForDevicesAsync() discovers no device.

### Crashlog

No exception is thrown.

## Configuration

**Version of the Plugin:** 2.1.1 and 2.2.0-pre2

**Platform:** Android 10

**Device:** Samsung Galaxy S10 and Google Pixel

**Android Target SDK**: 29+

travistodd099, 2020-09-06:
I am seeing the same.

samosim, 2020-09-06:
I have the same problem.

rafaelagp, 2020-09-09:
Solved it.

As of Android 6.0 Marshmallow, ACCESS_FINE_LOCATION permission is needed instead of ACCESS_COARSE_LOCATION for BLE functionality. Well, it isn't exactly needed, but for supporting all Android versions it is best.

closed 2020-09-09 by rafaelagp