dotnet-bluetooth-le/dotnet-bluetooth-le #455: BLE.Client.iOS crashed on my iPhone 10

douglaskbell, 2020-09-01:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Download latest PLUG.BLE FROM git
Latest VISUAL STUDIO 2019 Pro on Windows 10 and mac (latest OS)

2. VS > solution > right-click BLE.Client.iOS > Debug > Start

3.

## Expected behavior
Tell us what should happen  SHOULD RUN VIRGIN EXAMPLE

## Actual behavior
Tell us what happens instead   BRIEFLY SHOWED SPLASH, THEN BACK TO SPRINGBOARD

### Crashlog
[HotReload] (2020-09-01 11:49:14.3): WARN: (BLE.Client.iOS) Hot Reload disabled due to invalid linker settings
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Mono.Security.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/System.Core.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/System.Numerics.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/System.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/System.Drawing.Common.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Xamarin.iOS.dll [External]
Thread started:  #2
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/System.Xml.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Xamarin.Forms.Platform.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Xamarin.Forms.Core.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Microsoft.CSharp.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/MvvmCross.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/System.Net.Http.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/System.ServiceModel.Internals.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/System.Runtime.Serialization.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Xamarin.Forms.Platform.iOS.dll
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/MvvmCross.Forms.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/BTProgressHUD.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Acr.UserDialogs.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Plugin.BLE.Abstractions.dll
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Plugin.BLE.dll
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Plugin.Permissions.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Xamarin.Forms.Xaml.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Plugin.Settings.Abstractions.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/Plugin.Settings.dll [External]
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/BLE.Client.dll
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/MvvmCross.Plugins.BLE.iOS.dll
Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/BLEClientiOS.exe
2020-09-01 11:49:15.603 BLEClientiOS[3358:878268] 2020-09-01 11:49:15 [TRACE] (MvvmCross.Core.MvxSetup) Setup: Primary start
2020-09-01 11:49:15.603 BLEClientiOS[3358:878268] 2020-09-01 11:49:15 [TRACE] (MvvmCross.Core.MvxSetup) Setup: FirstChance start
2020-09-01 11:49:15.603 BLEClientiOS[3358:878268] 2020-09-01 11:49:15 [TRACE] (MvvmCross.Core.MvxSetup) Setup: PlatformServices start

Loaded assembly: /private/var/containers/Bundle/Application/2B4D0762-477A-49B4-9CA1-8580C6A8DCD1/BLEClientiOS.app/OpenTK-1.0.dll [External]
2020-09-01 11:49:15.690 BLEClientiOS[3358:878268] 2020-09-01 11:49:15 [TRACE] (MvvmCross.Core.MvxSetup) Setup: MvvmCross settings start

2020-09-01 11:49:15.690 BLEClientiOS[3358:878268] 2020-09-01 11:49:15 [TRACE] (MvvmCross.Core.MvxSetup) Setup: Singleton Cache start
2020-09-01 11:49:15.690 BLEClientiOS[3358:878268] 2020-09-01 11:49:15 [TRACE] (MvvmCross.Core.MvxSetup) Setup: ViewDispatcher start

Thread started: <Thread Pool> #3
Thread started: <Thread Pool> #4
Thread started: <Thread Pool> #5
Thread started: <Thread Pool> #6
2020-09-01 11:49:15.816 BLEClientiOS[3358:878320] 2020-09-01 11:49:15 [TRACE] (MvvmCross.Core.MvxSetup) Setup: Bootstrap actions

2020-09-01 11:49:15.822 BLEClientiOS[3358:878320] 2020-09-01 11:49:15 [TRACE] (MvvmCross.Core.MvxSetup) Setup: StringToTypeParser start

2020-09-01 11:49:15.826 BLEClientiOS[3358:878320] 2020-09-01 11:49:15 [TRACE] (MvvmCross.Core.MvxSetup) Setup: CommandHelper start

[... 69 lines cut ...]
	0x189083740 - /usr/lib/system/libsystem_pthread.dylib : start_wqthread

=================================================================
	Basic Fault Address Reporting
=================================================================
Memory around native instruction pointer (0x18915d7f4):
0x18915d7e4  c0 03 5f d6 c0 03 5f d6 30 41 80 d2 01 10 00 d4  .._..._.0A......

0x18915d7f4  e3 00 00 54 fd 7b bf a9 fd 03 00 91 b4 80 ff 97  ...T.{..........
0x18915d804  bf 03 00 91 fd 7b c1 a8 c0 03 5f d6 c0 03 5f d6  .....{...._...
_.
0x18915d814  d0 03 80 d2 01 10 00 d4 e3 00 00 54 fd 7b bf a9  ...........T.{..

The app has been terminated.
The app has been terminated.

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3      Plugin.BLE 2.1.1

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)  IOS 13

**Device:** e.g. HTC Sensation /i Phone 7 ...   IPHONE 10

janusw, 2021-04-09:
I can confirm this. I think the reason is that the BT permission request is missing. Similar things happen on Mac OS (Big Sur), see #498.

janusw, 2021-04-09:
Fixed via #499.

closed 2021-04-09 by janusw