dotnet-bluetooth-le/dotnet-bluetooth-le #456: Device Name is always Null

momone50, 2020-09-03:
Hi, I need to read Device Name from available devices, in order to filter results.
(My device has a specific device name, always beginning the same way).

Unfortunatly, device.Name is always empty. I think i should be able to read it before even connecting to it. Otherwise i couldn't
filter correctly

## Steps to reproduce

1. Scan for devices :
`await adapter.StartScanningForDevicesAsync();`

2.Add devices in a List in Discover Event Handler
`private void Device_Discovered_Event(object sender, Plugin.BLE.Abstractions.EventArgs.DeviceEventArgs e)
        {
            IDevice device = e.Device;
            devices.Add(device);                      
        }`

3.Try to read Device Name:
`foreach (IDevice device in devices)
                {
                    Debug.WriteLine(device.Name);
                }`

## Expected behavior
device.Name should contain a Name but it is always equal to "".
When scanning my device with a Ble Scan application, it manages to get the Name. With Plugin.BLE, it always fail.

**Version of the Plugin:** 2.1.1

**Platform:**  Android 5.0

**Device:** Samsung Grand Prime

StephanRaab, 2020-09-14:
So, you're sure it's not connected to something else and that's why it's not showing up?

momone50, 2020-09-14:
I don't have any problem to find the device. Only to retrieve its name. 
It's not connected to anything else. 
Actually, i found a workaround :
I first scan devices with my phone -> Phone manage to get name -> Then deviceName magically appears in my app...
Weird?

ske66, 2020-10-01:
There is the potential when scanning for devices that your device will pick up devices that don't have a Name. So it's best to filter these out in your list. I recommend making the following change to better filter out unwanted devices.

```
void DeviceDiscoveredEvent(object sender, DeviceEventArgs e)
{
    if (!devices.Contains(e.Device) && !String.IsNullOrEmpty(e.Device.Name) )
    devices.Add(e.Device);
}
```

momone50, 2020-10-01:
Hi,
I already tried to filter this way. I end up never finding my device doing so. It simply never manages to retrieve Name.
As i said in my previous post, it only works with the workaround i found

GO74, 2021-01-22:
Hi,
I'm using this plugin for quite a while and I now have the same problem as it seems.
Tested it on a S5 (lineage, Android 9) and Pixel 2 (Android 11) - Device.Name is always null.
I tried 2.1.1 and 2.1.0.

StephanRaab, 2021-01-25:
Another follow up question. The device you're trying to use, you're able to see the name using another application such as `nRF Connect`? Just want to make sure you're not accidentally using BLE, when you're wanting to be using Bluetooth Classic.

MynaBay, 2021-02-22:
I’m having the same issue. Device.name is always null on android.  Run the same Xamarin build on iOS and most names are filled in. My device’s name has never failed on iOS build, and never worked on Android.

janusw, 2021-04-10:
I definitely see devices that have a name via BLE.Client.Droid (on current master branch with Android 11). Not all devices may have their name set up, though (my scans also show several devices without a name). Simply seeing a device without a name is not necessarily a bug, I assume.

If you see the name of a particular device on iOS, but not on Android, that may indicate a problem on the Android side.

I'll note that Android 5 is rather old. Possibly this might be a problem with older Android versions? @MynaBay, which Android version are you using?

janusw, 2021-04-10:
> I definitely see devices that have a name via BLE.Client.Droid (on current master branch with Android 11).

I also see devices with a name on Android 8 (but that's the oldest Android version that I have access to right now).

condorkm, 2021-07-28:
I read this from stackoverflow:

I have an off-market Chinese tablet that has BLE support, however, it always return a BLE equipped device with its name as "null" when I call startLeScan. The issue was resolved by calling startDiscovery. Remember that if your app targets 23 or above, you will need to have location permissions for startDiscovery to work correctly. Hope it helps.

Could be there are some chinese tablets that are not returning name in statLeScan, and is required to user startDiscovery

condorkm, 2021-07-28:
Tested, if you put BluetoothAdapter.DefaultAdapter.StartDiscovery(); in your Mainactivity names are working