dotnet-bluetooth-le/dotnet-bluetooth-le #457: ICharateristic.WriteAsync > 20 bytes

daniel-streetec, 2020-09-18:
Hello,
I am encountering an Isse when Sending More than 20 bytes to an Characteristic with my Android Smartphone.
I have set the MTU Size to 256, which seems to work, because I can receive more than 20 bytes. It seems that, whenever sending more than 20 bytes, the data is being sent and shortly after, the device is disconnected from the phone.

[https://github.com/daniel-streetec/Plugin.BLE-WriteError](In this Repo) you can find the minimal example with source for ESP32 + Arduino

## Steps to reproduce

1. Connect to the BLE Device

2. Set MTU to > 20

3. Send more than 20 Bytes

## Expected behavior
Simply send all the data until MTU is reached

## Actual behavior
Nothing is send, device gets disconnected

### Crashlog

[0:] Sending 55 bytes
09-18 10:45:27.500 D/Mono    (29109): DllImport searching in: '__Internal' ('(null)').
09-18 10:45:27.500 D/Mono    (29109): Searching for 'java_interop_jnienv_new_byte_array'.
09-18 10:45:27.500 D/Mono    (29109): Probing 'java_interop_jnienv_new_byte_array'.
09-18 10:45:27.500 D/Mono    (29109): Found as 'java_interop_jnienv_new_byte_array'.
09-18 10:45:27.500 D/Mono    (29109): DllImport searching in: '__Internal' ('(null)').
09-18 10:45:27.500 D/Mono    (29109): Searching for 'java_interop_jnienv_set_byte_array_region'.
09-18 10:45:27.501 D/Mono    (29109): Probing 'java_interop_jnienv_set_byte_array_region'.
09-18 10:45:27.501 D/Mono    (29109): Found as 'java_interop_jnienv_set_byte_array_region'.
09-18 10:45:32.523 D/BluetoothGatt(29109): onClientConnectionState() - status=8 clientIf=11 device=FC:F5:C4:0E:43:FE
09-18 10:45:32.527 D/BluetoothGatt(29109): close()
09-18 10:45:32.534 D/BluetoothGatt(29109): unregisterApp() - mClientIf=11

## Configuration

**Version of the Plugin:** 2.1.1

**Platform:**  Android 10.0 API 29

**Device:**  Samsung Galaxy S20

memsom, 2021-01-18:
I had this issue. I think if you break up the data in to smaller chunks, it will probably work. However, I did this and still had issues. I actually ended up implementing reliable writes, and that mostly fixed the issue, so long as nothing else attempted to do anything whilst that was happening.

Unfortunately, implementing reliable writes will get you knee deep in library code. There is no easy way to do it without changing the source code. This is the code I used:

https://github.com/memsom/xamarin-bluetooth-le/blob/master/Source/Plugin.BLE.Android/WriteTransaction.cs

The key part is that it uses `gatt.ExecuteReliableWrite()`, and I think that is implemented in this commit:

https://github.com/memsom/xamarin-bluetooth-le/commit/ff94e918b61ebfac274a0f7c54bf37f0c1317a7e

memsom, 2021-01-27:
@daniel-streetec  Another thing - what model of phone are we talking about? We have recently had a bug with a Samsung S10, where setting the MTU to anything higher than 23 (that's 20 bytes of data, and 3 bytes for the header) would fail. This is due to a bug in the Samsung Bluetooth implementation in Android 10. We ended up blacklisting models we knew were problematic and setting the MTU to 23, otherwise we can set it to 517 (514 bytes of data and 3 bytes of header) and it will successfully transfer data using the mechanism I described above.

daniel-streetec, 2021-01-28:
Hi @memsom thanks for your answer.
I was trying with a Galaxy S20, so it could be the Samsung bug. I will try again with a different device, moreover I already switched from Arduino to ESP-IDF but that should not change the behavior as the ESP was accepting more than 20 bytes already.

I will get another phone and try again, but if it wont work on most Samsung phones I might need to stick to the MTU of 23 as I cannot ignore the Samsung users :)

memsom, 2021-02-02:
Well, I have a Galaxy S21 in hand, running Android 11, and the MTU bug is still happening in the Samsung Android. If we request anything above MTU of 23, it accepts it, but after data transfer starts, it completely kills the connection.

An update was just pushed by Samsung, so I can retest, but so far, no Samsung S series devices seem capable of supporting MTU > 23. I do have access to an A50, which might not be completely up to day. And this can request MTU of 517 and maintain a solid connection. I am tempted to make our app treat Samsung devices as crippled and restrict them all unless overridden in config.

lylerolleman, 2021-02-23:
The issue here is not exclusive to Samsung devices, I have also encountered it on Huawei and Xiaomi devices (granted both probably "inherited" the problem from Samsung)

It does not affect all Samsung devices however, as we had no issues on an A21. Appears to affect the SXX and Note devices

Thread on Samsung forums on the issue https://forum.developer.samsung.com/t/samsung-android-10-ble-connectivity-regression/509

Doesn't look like the problem is going to be fixed any time soon.

Due to the wide range of devices with the issue, our solution to the problem was to request the MTU increase, and then try writing a value at that larger MTU. If it fails and disconnects, we reconnect automatically, and just resign ourselves to running slowly.

memsom, 2021-06-03:
Yeah, we default to slow and override on devices we know work, but allow a config file to also set a different value.