dotnet-bluetooth-le/dotnet-bluetooth-le #459: Android 9 or higher don't starUpdateAsync

kvrscream, 2020-09-30:
Hello.  And sorry, my English is not good.  I'm trying to run StartUpdateAsync to notify the device and release a next measurement.
In android version 8 or lower it works and runs the command without problems. The same goes for the iPhone. But for any android version 9 or higher it just ignores the code.
How should I proceed

Steps to reproduce
1. Connect to the BLE Device 
2. Try  StartUpdateAsync  
Expected behavior
Update on indicate device.
Actual behavior
Skip rest code and stop get device info.
Crashlog
[BluetoothAdapter] STATE_ON
[BluetoothAdapter] BLE support array set: 010011
[BluetoothLeScanner] Start Scan with callback
[BluetoothLeScanner] onScannerRegistered() - status=0 scannerId=12 mScannerId=0
[InputMethodManager] HSIFW - flag : 0 Pid : 855
[Choreographer] Skipped 31 frames!  The application may be doing too much work on its main thread.
[inos.mais_amig] Accessing hidden field Landroid/view/animation/Animation;->mListener:Landroid/view/animation/Animation$AnimationListener; (greylist-max-p, reflection, allowed)
[Mono] Requesting loading reference 9 (of 21) of /storage/emulated/0/Android/data/com.kainos.mais_amigo/files/.__override__/Xamarin.Forms.Platform.Android.dll
[Mono] Loading reference 9 of /storage/emulated/0/Android/data/com.kainos.mais_amigo/files/.__override__/Xamarin.Forms.Platform.Android.dll asmctx DEFAULT, looking for Xamarin.Android.Support.v7.RecyclerView, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[Mono] Assembly Ref addref Xamarin.Forms.Platform.Android[0x7a55ae7000] -> Xamarin.Android.Support.v7.RecyclerView[0x7a55ae4700]: 2
[Mono] Requesting loading reference 3 (of 6) of /storage/emulated/0/Android/data/com.kainos.mais_amigo/files/.__override__/Xamarin.Android.Support.v7.RecyclerView.dll
[Mono] Loading reference 3 of /storage/emulated/0/Android/data/com.kainos.mais_amigo/files/.__override__/Xamarin.Android.Support.v7.RecyclerView.dll asmctx DEFAULT, looking for Xamarin.Android.Support.Compat, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[Mono] Assembly Ref addref Xamarin.Android.Support.v7.RecyclerView[0x7a55ae4700] -> Xamarin.Android.Support.Compat[0x7a5e5d2f80]: 6
[Mono] Requesting loading reference 2 (of 6) of /storage/emulated/0/Android/data/com.kainos.mais_amigo/files/.__override__/Xamarin.Android.Support.v7.RecyclerView.dll
[Mono] Assembly Ref addref Xamarin.Android.Support.v7.RecyclerView[0x7a55ae4700] -> Java.Interop[0x7a55ae7580]: 12
[Mono] Requesting loading reference 4 (of 6) of /storage/emulated/0/Android/data/com.kainos.mais_amigo/files/.__override__/Xamarin.Android.Support.v7.RecyclerView.dll
[Mono] Loading reference 4 of /storage/emulated/0/Android/data/com.kainos.mais_amigo/files/.__override__/Xamarin.Android.Support.v7.RecyclerView.dll asmctx DEFAULT, looking for Xamarin.Android.Support.CustomView, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[Mono] Assembly Ref addref Xamarin.Android.Support.v7.RecyclerView[0x7a55ae4700] -> Xamarin.Android.Support.CustomView[0x7a55ad5680]: 2
[Choreographer] Skipped 41 frames!  The application may be doing too much work on its main thread.
[Mono] DllImport searching in: '__Internal' ('(null)').
[Mono] Searching for 'java_interop_jnienv_get_byte_array_region'.
[Mono] Probing 'java_interop_jnienv_get_byte_array_region'.
[Mono] Found as 'java_interop_jnienv_get_byte_array_region'.
[ViewRootImpl@c75a8a1[MainActivity]] ViewPostIme pointer 0
[ViewRootImpl@c75a8a1[MainActivity]] ViewPostIme pointer 1
[inos.mais_amig] Explicit concurrent copying GC freed 9155(836KB) AllocSpace objects, 3(60KB) LOS objects, 49% free, 2527KB/5054KB, paused 93us total 26.482ms
[Mono] GC_TAR_BRIDGE bridges 576 objects 9061 opaque 2496 colors 576 colors-bridged 576 colors-visible 576 xref 52 cache-hit 0 cache-semihit 0 cache-miss 0 setup 0.35ms tarjan 15.70ms scc-setup 0.26ms gather-xref 0.03ms xref-setup 0.03ms cleanup 1.04ms
[Mono] GC_BRIDGE: Complete, was running for 33.91ms
[Mono] GC_MINOR: (Nursery full) time 56.83ms, stw 58.23ms promoted 1705K major size: 2800K in use: 2062K los size: 1024K in use: 796K
[Mono] Requesting loading reference 1 (of 2) of Java.Interop.dll
[Mono] Loading reference 1 of Java.Interop.dll asmctx DEFAULT, looking for System.Core, Version=2.0.5.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e
[Mono] Assembly Ref addref Java.Interop[0x7a55ae7580] -> System.Core[0x79eee5c380]: 7
[BluetoothAdapter] STATE_ON
[BluetoothAdapter] STATE_ON
[BluetoothLeScanner] Stop Scan with callback
[BluetoothAdapter] STATE_ON
[BluetoothGatt] connect() - device: E1:EB:0A:09:B9:21, auto: false
[BluetoothAdapter] isSecureModeEnabled
[BluetoothGatt] registerApp()
[BluetoothGatt] registerApp() - UUID=5518910b-36a7-4838-948e-199b5240e4d0
[BluetoothGatt] onClientRegistered() - status=0 clientIf=12
[BluetoothGatt] onClientConnectionState() - status=0 clientIf=12 device=E1:EB:0A:09:B9:21
Thread started:  #10
[BluetoothGatt] discoverServices() - device: E1:EB:0A:09:B9:21
[BluetoothGatt] onConnectionUpdated() - Device=E1:EB:0A:09:B9:21 interval=6 latency=0 timeout=500 status=0
[BluetoothGatt] onSearchComplete() = Device=E1:EB:0A:09:B9:21 Status=0
[Mono] DllImport searching in: '__Internal' ('(null)').
[Mono] Searching for 'java_interop_jnienv_call_int_method'.
[Mono] Probing 'java_interop_jnienv_call_int_method'.
[Mono] Found as 'java_interop_jnienv_call_int_method'.
[BluetoothGatt] onConnectionUpdated() - Device=E1:EB:0A:09:B9:21 interval=36 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=E1:EB:0A:09:B9:21 interval=792 latency=0 timeout=400 status=0
[BluetoothGatt] setCharacteristicNotification() - uuid: 00002a1c-0000-1000-8000-00805f9b34fb enable: true
[Mono] DllImport searching in: '__Internal' ('(null)').
[Mono] Searching for 'java_interop_jnienv_new_byte_array'.
[Mono] Probing 'java_interop_jnienv_new_byte_array'.
[Mono] Found as 'java_interop_jnienv_new_byte_array'.
[Mono] DllImport searching in: '__Internal' ('(null)').
[Mono] Searching for 'java_interop_jnienv_set_byte_array_region'.
[Mono] Probing 'java_interop_jnienv_set_byte_array_region'.
[Mono] Found as 'java_interop_jnienv_set_byte_array_region'.
Configuration
Version of the Plugin: 2.1.1
Platform: Android 10.0 API 29
Device: Samsung A20, Moto G7

kvrscream, 2020-09-30:
@smsissuechecker  Sorry. I edited for you're.

ske66, 2020-10-01:
That doesn't look like a very useful Crashlog. Can you share a code example as well as output/exceptions?

kvrscream, 2020-10-01:
Hi @ske66 
Look my code here:

`if (device != null)
                {

                    await adapter.StopScanningForDevicesAsync();
                    await adapter.DisconnectDeviceAsync(device);
                    await adapter.ConnectToDeviceAsync(device);

                    var services = await device.GetServicesAsync();

                    ICharacteristic characteristicTemperature = null;
                    ICharacteristic characteristicBattery = null;

                    foreach (var service in services)
                    {
                        var caracteristicas = await service.GetCharacteristicsAsync();

                        foreach (var carac in caracteristicas)
                        {
                            //Bug android 9 ou > -- Problemas ao notificar o device
                            //Checa se é Bateria ou Temperatura
                            if (carac.Uuid == "2a19" || carac.Uuid == "2a1c"
                                || carac.Name.ToLower().Equals("temperature measurement")
                                || carac.Name.ToLower().Equals("battery level"))
                            {

                                if(carac.Name.ToLower().Equals("battery level") || carac.Uuid == "2a19")
                                {
                                    characteristicBattery = carac;
                                }

                                if(carac.Name.ToLower().Equals("temperature measurement") || carac.Uuid == "2a1c")
                                {
                                    characteristicTemperature = carac;
                                }

                            }
                        }
                    }

                   
                        characteristicTemperature.ValueUpdated += (o, args) =>
                        {
                            byte[] bytes = args.Characteristic.Value;
                        };
                        
                        await characteristicTemperature.StartUpdatesAsync();

                    `

kvrscream, 2021-02-23:
Hi everyone. I found the solution \o/

The problem is on Android device. I reseted my bluetooth and network setting and my Android device and now startUpdateAsync normaly.

Thanks all for help me.

closed 2021-02-23 by kvrscream