dotnet-bluetooth-le/dotnet-bluetooth-le #461: Cant notify my device on propertie is Indicate Android 10

kvrscream, 2020-10-02:
I have some problems to use Plugin.BLE on Android 10.
I can't await characterist.StartUpdateAsync(). All time when I call this method my code skip the rest.
My characterist has propertie Indicate.

My code: 
`
await adapter.StopScanningForDevicesAsync();
                    //await adapter.DisconnectDeviceAsync(device);
                    await adapter.ConnectToDeviceAsync(device);
                    
                    var services = await device.GetServicesAsync();
                    
                    ICharacteristic characteristicTemperature = null;
                    ICharacteristic characteristicBattery = null;

                    foreach (var service in services)
                    {
                        var caracteristicas = await service.GetCharacteristicsAsync();

                        foreach (var carac in caracteristicas)
                        {
                            
                            var descritors = await carac.GetDescriptorsAsync();

                            //Bug android 9 ou > -- Problemas ao notificar o device
                            //Checa se é Bateria ou Temperatura
                            if (carac.Uuid == "2a19" || carac.Uuid == "2a1c"
                                || carac.Name.ToLower().Equals("temperature measurement")
                                || carac.Name.ToLower().Equals("battery level"))
                            {

                                if (carac.Name.ToLower().Equals("battery level") || carac.Uuid == "2a19")
                                {
                                    characteristicBattery = carac;
                                }

                                if (carac.Name.ToLower().Equals("temperature measurement") || carac.Uuid == "2a1c")
                                {
                                    characteristicTemperature = carac;
                                }

                            }
                        }
                    }

                    if (characteristicTemperature.CanUpdate)
                    {

                        var descriptor = await characteristicTemperature.GetDescriptorsAsync();

                        characteristicTemperature.ValueUpdated += (o, args) =>
                        {
                            byte[] bytes = args.Characteristic.Value;
                        };

                        await characteristicTemperature.StartUpdatesAsync();

                    }`

In iOS and Android 8 or bottom this method runs ok.

## Steps to reproduce

1.  Conect to device find service and characterist

2. Try run await characterist.StartUpdateAsync()

## Expected behavior
Continue rest of code and get value from device.

## Actual behavior
Skip rest of code and don't notify device.
### Crashlog

[BluetoothGatt] discoverServices() - device: E1:EB:0A:09:B9:21
[BluetoothGatt] onConnectionUpdated() - Device=E1:EB:0A:09:B9:21 interval=6 latency=0 timeout=500 status=0
[BluetoothGatt] onSearchComplete() = Device=E1:EB:0A:09:B9:21 Status=0
[Mono] DllImport searching in: '__Internal' ('(null)').
[Mono] Searching for 'java_interop_jnienv_call_int_method'.
[Mono] Probing 'java_interop_jnienv_call_int_method'.
[Mono] Found as 'java_interop_jnienv_call_int_method'.

## Configuration

**Version of the Plugin:** 2.1.1

**Platform:** Android 10

**Device:** Samsung A20

kvrscream, 2020-10-06:
Hi everyone.
I try in other devices with android 10 and 9. I believe the problem is on Samsung no in android.

If someone can help me I thank.

MelissaW990, 2020-10-06:
I have also encountered this issue and believe it is a problem with Android 10 on Samsung devices only. I found this forum which describes the issue in more detail, seems Samsung have been aware of the bug since the start of the year but no fix has been put in place yet. Really poor on their behalf! https://forum.developer.samsung.com/t/samsung-android-10-ble-connectivity-regression/509/92

kvrscream, 2020-10-06:
@MelissaW990 
Thanks so much for help me.
Is a really big problem for me.

Farin94, 2020-12-25:
I just had the same issue on a Xiaomi Redmi Note 7 device.
Seems to me that this issue is not limited to just Samsung devices...

kvrscream, 2021-02-24:
Hi everyone.

I found it a solution.
I reset connection and bluetooth setting for my android device and the application work normality.

Thanks everyone for help me.

closed 2021-02-24 by kvrscream